chipsalliance/chisel #443: eliminate redundant wires/copies

sols1, 2017-01-19:
Chisel generated RTL has a lot copies, e.g.:

```
  assign RRTestCombinational_79_1_io_rready = _T_1512;
  assign RRTestCombinational_79_1_io_wvalid = _T_1515;
  assign RRTestCombinational_79_1_io_wdata = _T_604;
  assign _T_168_ready = _T_2351;
  assign _T_168_valid = _T_3492;
  assign _T_168_bits_read = _T_3484;
  assign _T_168_bits_index = _T_3486;
  assign _T_168_bits_data = io_in_0_hwdata;
  assign _T_168_bits_mask = _T_3488;
  assign _T_168_bits_extra = 1'h0;
  assign _T_202_ready = 1'h1;
  assign _T_202_valid = _T_2354;
  assign _T_202_bits_read = _T_238_bits_read;
  assign _T_202_bits_data = _T_3478;
  assign _T_202_bits_extra = _T_238_bits_extra;
  assign _T_238_ready = _T_2353;
  assign _T_238_valid = _T_2352;
  assign _T_238_bits_read = _T_168_bits_read;
  assign _T_238_bits_index = _T_168_bits_index;
  assign _T_238_bits_data = _T_168_bits_data;
  assign _T_238_bits_mask = _T_168_bits_mask;
  assign _T_238_bits_extra = _T_168_bits_extra;
```

This makes RTL larger (slower to compile, simulate with waveform, etc.), more difficult to debug RTL (e.g. trace signal drivers in DVE).

Also, code coverage metrics can be skewed (not sure how coverage tools deal with this).

It should be possible to replace module-internal copies, .e.g. replace `_T_2352` with `_T_238_valid` in the code above and get rid of `_T_2352`.

`_T_2352` can be put in comments if it is needed for whatever reason (e.g. debugging):

```
  wire _T_238_valid; // _T_2352
```

sols1, 2017-01-19:
Another suggestion is to generate wire assignments to reduce generated code size, i.e.:

```
  wire AB = A & B;
  wire CD = C & D;
  wire O = AB | CD;
  wire F = ~O;
```

instead of:

```
  wire F;
  wire AB, CD, O;
  assign AB = A & B;
  assign CD = C & D;
  assign O = AB | CD;
  assign F = ~O;
```

ducky64, 2017-02-06:
Can you submit the chisel code that generates this?

shunshou, 2017-02-06:
^ It's not a bug, b/c the output is legal but it's really bad QOR. I'm pretty sure almost all sufficiently large Chisel code generates something like this (i.e. uses assign [nothing to do w/ large designs; would minimize # of lines if you just did the wire thing] + "complicated" optimizations seem to throw in random gen_1 = gen_2 blah blah). Makes visual inspection (for the purposes of debugging anything -- even if you say it's not the point to have readable Verilog) a PITA, and if it's true that it significantly affects tool run times/coverage, that's even worse! I imagine it's probably one of those annoying optimization things that people didn't really want to do, b/c it's just... really annoying and doesn't hurt correctness.

sols1, 2017-02-07:
Yes, I agree with @shunshou - it is not a bug - it is an obvious optimization.

Take any large Chisel RTL, generate Verilog and you'll find plenty of examples of useless _T_ wire copies.

shunshou, 2017-02-07:
In that regard, Chisel3 hasn't changed very much from Chisel2 ;)

ducky64, 2017-04-18:
Actually a firrtl issue, ucb-bar/firrtl#414

closed 2017-04-18 by ducky64