chipsalliance/chisel #445: Shouldn't Vec(Reg()) generate an error?

ucbjrl, 2017-01-19:
This has been deprecated in Chisel2. It should cause an error in Chisel3. It seems to generate "incorrect" code.

aswaterman, 2017-01-20:
Vec(reg1, reg2, reg3, reg4) is well-defined and shouldn't be illegal.  It's just that it's read-only.

Vec(n, Reg(t)) is useless and should generate an error.

ducky64, 2017-01-20:
I think Vec(Reg()) will be an error from https://github.com/ucb-bar/chisel3/issues/368, since Reg will return a Bound type while Vec expects an unbound type.
Vec(Reg().getType) will be legal and provides the same functionality, but more explicit.

sdtwigg, 2017-01-20:
Vec(Reg(...)) is correct. Vec(Reg().getType) would actually be incorrect. Both are using the apply(T*) overload, which has the contract of taking hardware and returning a dynamically indexable 'read-only' Wire(Vec(...)).

ducky64, 2017-01-20:
I think we need to clarify whether Vec returns a synthesizable node (bound type) or data model node (unbound type). Otherwise it gets confusing.
Or the third option of "either, depending on what is passed in (as long as it's all consistent)", which might be more powerful but is less intuitive.

sdtwigg, 2017-01-20:
Right so, I've already rambled on the particulars of Vec here https://github.com/ucb-bar/chisel3/issues/336#issuecomment-255287979

but the general pushback has been that such cleanups are not backwards compatible, so you need to actually decide on whether you want backwards compatibility or consistency here (for just the 'compatibility layer' or the entire API) since the old API was simple not 'consistent'.

ducky64, 2017-01-20:
I think a possible implementation to limit the scope of types (bound vs. unbound) can be done with Bindings (and with the proposal in #440). Binding checks can be ignored in chisel2 compatibility mode, and this is probably sufficiently unintuitive behavior that it would be a good idea to update the chisel3 code that uses unsafe constructs. Or (not recommended) set Chisel in NotStrict mode.

ducky64, 2017-01-26:
Resolution from today's meeting:
- Vec(chiselType) will take a Chisel Type create a Chisel Type.
- Vec(args*), Vec(elts) will be deprecated and replaced with Vec.wire, making the difference expicit between whether you're creating Hardware or a Chisel Type.

Example:
```scala
val myVec = Reg(Vec(2, UInt(...)))  // either this, Reg of Vec of two UInts
val myVec = Vec.wire(Reg(UInt(...)), Reg(UInt(...)))  // ... or this, Vec Wire of two UInt Regs
myVec := Vec.wire(5.U, 4.U)  // will work with this
```

ccelio, 2017-01-27:
Hey guys, incoming bike-shedding: 

`Vec.wire` seems okay-ish, but a little weird.  I'm struggling to recall an example elsewhere in Chisel where we call a function on an object in this manner. For example:

> `RegNext`, `RegEnable`, `RegInit`. 

Perhaps, to maintain symmetry, we should instead have `Reg.next(...)` and friends?  Or `VecWire()` / `genVec()` objects? I dunno, just a thought.

ducky64, 2017-01-27:
Good point, we definitely should be consistent.

From an implementation standpoint, we can deprecate RegNext/Enable/Init and move it to the compatibility layer if we wanted to move towards Reg.next / etc.

Side note: in #455, I have a separate Reg.init that takes in both a type and an initializer (so type is not auto derived) as an internal function (init has to be declared with the reg due to FIRRTL semantics). This can of course be renamed, but it also serves as a bit of a differentiator between RegNext/RegInit which auto-derives their chisel types.

azidar, 2017-08-21:
Fixed by #635

closed 2017-08-21 by azidar