chipsalliance/chisel #446: Is there a reason we do an explicit Mux compatibility check?

ducky64, 2017-01-20:
Bits.scala:788
```scala
  private[core] def typesCompatible[T <: Data](x: T, y: T): Boolean = {
    val sameTypes = x.getClass == y.getClass
    val sameElements = x.flatten zip y.flatten forall { case (a, b) => a.getClass == b.getClass && a.width == b.width }
    val sameNumElements = x.flatten.size == y.flatten.size
    sameTypes && sameElements && sameNumElements
  }

  private def doAggregateMux[T <: Data](cond: Bool, con: T, alt: T)(implicit sourceInfo: SourceInfo): T = {
    require(typesCompatible(con, alt), s"can't Mux between heterogeneous types ${con.getClass} and ${alt.getClass}")
    doMux(cond, con, alt)
  }
```
Is there a reason we have an explicit typesCompatible check, instead of just using connect semantics? I'm trying (once again) to refactor flatten/elements out of Data, because it really doesn't belong there.

sdtwigg, 2017-01-20:
Mux has to return (a new) something and the current implementation has to pick either con or alt to clone so it is simpler to explain mechanics if that arbitrary choice 'doesn't matter'. 

Side note: Why is this in Bits.scala....

ducky64, 2017-01-20:
I can see why the type check is necessary now, but the element check seems like a case of bad abstraction boundaries (why does Mux get access to flatten?). It might make more sense to have a Aggregate.compatible(that) function (and something similar is probably already invoked in a strict connect check anyways).

Yeah, I think I'll try to move Mux out of Bits.scala.

ducky64, 2017-04-11:
Made consistent with everything else with the new clone supertype helpers.

closed 2017-04-11 by ducky64