chipsalliance/chisel #449: Iterable or Seq

ducky64, 2017-01-24:
We have uses of both (and a few instances of IndexedSeq, for kicks and giggles) spread across our codebase and API. We should standardize on one.

aswaterman, 2017-01-24:
I don't think so.  They have different contracts.  Iterable makes sense to use when the traversal order doesn't affect semantics; Seq makes sense when it does.  It's actually good self-documentation.

Mux1H's overloads correctly use Iterable in one case and Seq in the others.

aswaterman, 2017-01-24:
I can see the argument for replacing the few uses of IndexedSeq with Seq.

ducky64, 2017-02-06:
Resolution: each has their own contract and we're going with least-specific class. May consider individual instances that don't make sense.

closed 2017-02-06 by ducky64