chipsalliance/chisel #452: Stop dumping test artifacts in chisel3/

ducky64, 2017-01-24:
I'm looking at you, `TopOfDiamond`, `UsesMuchUsedModule`, and `DummyModule`. We have a standard convention for test directories (currently dumped in `/tmp` as per `TesterDriver.execute(...)`, though it wouldn't necessarily be a bad idea to change this to something like `build/`).

Practically this makes `git status` a lot more messy as you have random test artifacts in the untracked files section.

Some refactoring of TesterDriver.scala to have incrementally more defined tester Drivers (for example, a basic one with just destination directory, one on top with the Verilator compilation, and one on top of that with the BasicTester stuff, etc) might be a good solution here.

jackkoenig, 2017-01-24:
We definitely need to stop using `/tmp` as well. I agree with switching to `build/` or something, and we should consider deleting successful tests (although expecting the user to be able to nuke `build/` doesn't seem to unreasonable

ducky64, 2017-01-24:
Perhaps have a new test clear the results / artifacts of the previous test? The random string of numbers after the test name (like `LastAssignTester1352679602163862897`) isn't helpful at all.

There's definitely an argument for keeping artifacts indefinitely, though I don't think that's how most testing systems work. But if you really wanted to keep all test results, at least make the numbers meaningful, like a seconds-resolution timestamp.

jackkoenig, 2017-01-24:
Clearing the previous results seems reasonable. If we do it this way, we can just drop the random string of numbers, those are generated by Java when you make a temp directory. Alternatively we could do timestamps, I could go either way.

chick, 2017-01-24:
I'm all for cleaning these up but cleaning the results as a test starts isn't going to work because scalatest runs things in parallel.  
Seems like a good first step would push these things into a directory, *test_run_dir* is what chisel_testers uses, the entire directory can be wiped easily then

ducky64, 2017-01-24:
Tests could run in their own directory (like `build/LastAssignTester/`) which can be wiped independently of other tests. Though there's still the problem is multiple tests (or even the same test with different parameters) under the same Spec, which might cause a collision. So it might just have to be don't wipe anything and timestamp things. Or provide an API to disambiguate different specs / tests, which would be more ideal but also cumbersome. Maybe ScalaTest has something to better automate this?

chick, 2017-01-24:
Multiple tests stomping on each other may be the cause of this report by @shunshou here [Verilator backend breaks for multiple tests](https://github.com/ucb-bar/chisel-testers/issues/87)

commit 06857f16 (2017-01-30): Use new createTestDirectory. Fixes #452.
commit 24ad38ab (2017-02-01): Move backend compilation utilities (#400)

* Move copyResourceToFile() to BackendCompilationUtilities.
* Move BackendCompilationUtilities into a firrtl util package.
Some of this could be moved into a more general tools package, but since chisel3 already has a dependency on firrtl ...
* Push util down into firrtl so as not to conflict with scala.util.
* Use new createTestDirectory. Fixes #452.
closed 2017-02-01 by jackkoenig