chipsalliance/chisel #454: Bundles not working properly?

shunshou, 2017-01-25:
Can anyone see if there's anything wrong with the code here? https://github.com/ucb-art/Chisel3DSPExample/blob/master/src/test/scala/UsefulExamples/SimpleTBwGenTypeOption.scala

Relevant code:

```
class DataTypeBundle[R <: Data:Real](genType: R, width: Width, binaryPoint: BinaryPoint) extends Bundle {
  val gen = genType.chiselCloneType
  val s = SInt(width)
  val f = FixedPoint(width, binaryPoint)
  val u = UInt(width)
  override def cloneType: this.type = new DataTypeBundle(gen, width, binaryPoint).asInstanceOf[this.type]
}

class Interface[R <: Data:Real](genShort: R, genLong: R, includeR: Boolean, p: TestParams) extends Bundle {

  val smallW = p.smallW.W
  val bigW = p.bigW.W
  val smallBP = p.smallBP.BP
  val bigBP = p.bigBP.BP
  val vecLen = p.vecLen

  val r = if (includeR) Some(DspReal()) else None
  val b = Bool()
  val cGenL = DspComplex(genLong)
  val cFS = DspComplex(FixedPoint(bigW, smallBP))
  val cR = DspComplex(DspReal())

  val short = new DataTypeBundle(genShort, smallW, smallBP)
  val long = new DataTypeBundle(genLong, bigW, bigBP)

  val vU = Vec(vecLen, UInt(smallW))
  val vS = Vec(vecLen, SInt(smallW))  
  
  override def cloneType: this.type = new Interface(genShort, genLong, includeR, p).asInstanceOf[this.type]
}
```

TestParams is a case class.

The IO inside my module looks like this:

```
 val io = IO(new Bundle {
    val i = Input(new Interface(genShort, genLong, includeR, p))
    val o = Output(new Interface(genShort, genLong, includeR, p)) }) 
```

The particular reflection error is this:

```
[info]   Cause: java.lang.reflect.InvocationTargetException:
[info]   at sun.reflect.GeneratedMethodAccessor8.invoke(Unknown Source)
[info]   at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[info]   at java.lang.reflect.Method.invoke(Method.java:606)
[info]   at chisel3.core.Bundle.chisel3$core$Bundle$$getBundleField(Aggregate.scala:443)
[info]   at chisel3.core.Bundle$$anonfun$elements$1.apply(Aggregate.scala:428)
[info]   at chisel3.core.Bundle$$anonfun$elements$1.apply(Aggregate.scala:427)
[info]   at scala.collection.IndexedSeqOptimized$class.foreach(IndexedSeqOptimized.scala:33)
[info]   at scala.collection.mutable.WrappedArray.foreach(WrappedArray.scala:35)
[info]   at chisel3.core.Bundle.elements$lzycompute(Aggregate.scala:427)
[info]   at chisel3.core.Bundle.elements(Aggregate.scala:424)
```

shunshou, 2017-01-25:
OK, @sdtwigg helped me debug (and might PR a change later?) It was unhappy that I had a parameter called "width"... getPublicFields of the bundle resulted in

```
WrappedArray(public chisel3.core.FixedPoint SimpleTB.DataTypeBundle.f(), public chisel3.core.SInt SimpleTB.DataTypeBundle.s(), public chisel3.core.UInt SimpleTB.DataTypeBundle.u(), public chisel3.internal.firrtl.Width chisel3.core.Aggregate.width())
```
Last line, like what? -_-

ducky64, 2017-04-12:
Fix PR is merged.

closed 2017-04-12 by ducky64