chipsalliance/chisel #457: SeqMem Clks

shunshou, 2017-01-26:
Just double checking, since it doesn't seem to be anywhere in Chisel3. Although FIRRTL supports SeqMem's having multiple clks, Chisel doesn't? Is it just because this hasn't been urgent? Also, I haven't tried yet, but I assume changing the Module clk would change both the read and write clocks to not use the default 'clock'?

ben-k, 2017-01-26:
I believe you are correct on both counts.

aswaterman, 2017-01-26:
The goal is for the forthcoming withClockAndReset construct to govern the clock inherited by a port declaration, rather than associating a clock with the memory itself.

ben-k, 2017-01-26:
Agreed, but is there a vision for how this would work with distinct read and write clocks for a SeqMem?

aswaterman, 2017-01-26:
Yeah, that's what I mean.  You declare the Mem outside, then put the ports inside two different withClockAndReset blocks.

ducky64, 2017-04-18:
Should be fixed by withClockAndReset

closed 2017-04-18 by ducky64