chipsalliance/chisel #462: Inheritance of Modules with io declared in the base class

jchang0, 2017-01-27:
I have an abstract class with IO declared:

class AbstractCacheModelParameterIO(num_addr_mon: Int, address_width: Int) extends Bundle {
  val addr_mon = Vec(num_addr_mon, UInt(address_width.W))
  override def cloneType: this.type = new AbstractCacheModelParameterIO(num_addr_mon, address_width).asInstanceOf[this.type]
}

abstract class BaseClass extends Module {
  def proc_func: (AbstractCacheModelParameterIO) => Unit

   val io = IO(new Bundle {
    val param_io = Input(new AbstractCacheModelParameterIO(num_addr_mon, 16))
  })
  this.proc_func(io.param_io)
}

Then extends it that access it's io
class ExtendClass extends BaseClass {
  def proc_func: (TLBundle, AbstractCacheModelParameterIO) => Unit = 
      (param_io: AbstractCacheModelParameterIO) => {

  val test_wire = Wire(UInt(16.W))
  test_wire := io.param_io.addr_mon[0]
}

Then instantiate it and make the connections:
val inst = Module(new ExtendClass)
inst.io.param_io = Vec(Seq.fill(num_addr_mon)(10.U))

The firrtl it generates references Extended Class:
  ExtendClass.io.param_io.addr_mon[0]

However this signal isn't declared (it's declared as BaseClass.io.param_io.addr_mon[0]) and it's giving me firrtl pass error.

I'm not sure if inherited IO is supposed to work.  Maybe it's some funky interaction with overriding proc_func and passing the param_io.  Or maybe it's something else happening (the actual code actually references the needed io.param_io signal into another module that got reduced to a wire.  But not sure if this usage scenario is a known limitation.

jchang0, 2017-01-27:
Upon further analysis, the badly generated firrtl is not due to inheritance, I think.  May due to something else that I'm doing.  Will open another ticket if I narrow it down and remains an issue.

closed 2017-01-27 by jchang0

ducky64, 2017-01-27:
Currently, we don't support multiple IO in a module (the only IO must be `.io`). This will be fixed in the very near future.
Also, there's a difference between `Input(...)` and `Output(...)` (which sets directionality) and `IO(...)` (which marks a chisel type as a module-level IO).