chipsalliance/chisel #465: Chisel exception during elaboration on use before declaration

sols1, 2017-01-28:
Source code (I2C.scala:90):
```
  val sclSync   = dSCL && !sSCL && io.port.scl.oe
```

`dSCL` and `sSCL` are declared later:

```
   val sSCL = (new Majority(fSCL.toBools.toSet)).out

   val dSCL = Reg(init = Bool(true), next = sSCL)
```

Exception:

```
[info] [0.003] Elaborating design...
[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:193)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:193)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:218)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:216)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:216)
	at chisel3.Driver$.elaborate(Driver.scala:193)
	at util.HasGeneratorUtilities$class.elaborate(GeneratorUtils.scala:50)
	at sifive.enterprise.generator.EnterpriseGenerator$.elaborate(Generator.scala:40)
	at util.GeneratorApp$class.circuit(GeneratorUtils.scala:82)
	at sifive.enterprise.generator.EnterpriseGenerator$.circuit$lzycompute(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator$.circuit(Generator.scala:40)
	at util.GeneratorApp$class.generateFirrtl(GeneratorUtils.scala:88)
	at sifive.enterprise.generator.EnterpriseGenerator$.generateFirrtl(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator$.delayedEndpoint$sifive$enterprise$generator$EnterpriseGenerator$1(Generator.scala:42)
	at sifive.enterprise.generator.EnterpriseGenerator$delayedInit$body.apply(Generator.scala:40)
	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
	at scala.App$class.main(App.scala:76)
	at sifive.enterprise.generator.EnterpriseGenerator$.main(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator.main(Generator.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
Caused by: java.lang.NullPointerException
	at sifive.blocks.devices.i2c.I2CModule$class.$init$(I2C.scala:90)
	at sifive.blocks.devices.i2c.TLI2C$$anonfun$15$$anon$2.<init>(I2C.scala:174)
	at sifive.blocks.devices.i2c.TLI2C$$anonfun$15.apply(I2C.scala:174)
	at sifive.blocks.devices.i2c.TLI2C$$anonfun$15.apply(I2C.scala:174)
	at uncore.tilelink2.TLRegisterRouter.module$lzycompute(RegisterRouter.scala:118)
	at uncore.tilelink2.TLRegisterRouter.module(RegisterRouter.scala:118)
	at diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:55)
	at diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:55)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:55)
	at diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:52)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at diplomacy.LazyModule.instantiate(LazyModule.scala:52)
	at diplomacy.LazyModuleImp.<init>(LazyModule.scala:133)
	at rocketchip.BareTopModule.<init>(BaseTop.scala:24)
	at rocketchip.BaseTopModule.<init>(BaseTop.scala:63)
	at sifive.enterprise.everywhere.guava.GuavaRocketChipTopModule.<init>(Top.scala:110)
	at sifive.enterprise.everywhere.guava.GuavaRocketChipTop.module$lzycompute(Top.scala:87)
	at sifive.enterprise.everywhere.guava.GuavaRocketChipTop.module(Top.scala:87)
	at sifive.enterprise.everywhere.guava.GuavaTop$$anonfun$2.apply(Top.scala:152)
	at sifive.enterprise.everywhere.guava.GuavaTop$$anonfun$2.apply(Top.scala:152)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at sifive.enterprise.everywhere.guava.GuavaTop.<init>(Top.scala:152)
	at sifive.enterprise.everywhere.guava.TestHarness$$anonfun$8.apply(TestHarness.scala:61)
	at sifive.enterprise.everywhere.guava.TestHarness$$anonfun$8.apply(TestHarness.scala:61)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at sifive.enterprise.everywhere.guava.TestHarness.<init>(TestHarness.scala:61)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:193)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:193)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:218)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:216)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:216)
	at chisel3.Driver$.elaborate(Driver.scala:193)
	at util.HasGeneratorUtilities$class.elaborate(GeneratorUtils.scala:50)
	at sifive.enterprise.generator.EnterpriseGenerator$.elaborate(Generator.scala:40)
	at util.GeneratorApp$class.circuit(GeneratorUtils.scala:82)
	at sifive.enterprise.generator.EnterpriseGenerator$.circuit$lzycompute(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator$.circuit(Generator.scala:40)
	at util.GeneratorApp$class.generateFirrtl(GeneratorUtils.scala:88)
	at sifive.enterprise.generator.EnterpriseGenerator$.generateFirrtl(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator$.delayedEndpoint$sifive$enterprise$generator$EnterpriseGenerator$1(Generator.scala:42)
	at sifive.enterprise.generator.EnterpriseGenerator$delayedInit$body.apply(Generator.scala:40)
	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
	at scala.App$class.main(App.scala:76)
	at sifive.enterprise.generator.EnterpriseGenerator$.main(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator.main(Generator.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
[trace] Stack trace suppressed: run last federation/compile:runMain for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last federation/compile:runMain for the full output.
[error] (federation/compile:runMain) Nonzero exit code: 1
[error] Total time: 9 s, completed Jan 27, 2017 4:19:31 PM
```

shunshou, 2017-01-28:
Aside from better error messaging, I don't think the intention was to ever support declaration after use... At least that was definitely the case in Chisel2...

jackkoenig, 2017-01-28:
Any exception that is not a Chisel exception is a bug. Thanks for the report.

jackkoenig, 2017-01-28:
Well any that is due to Chisel, this might not be possible to deal with actually.

sdtwigg, 2017-01-28:
For this localized case, the compiler should have issued a warning (assuming 2.11+)

ducky64, 2017-02-06:
Can't be solved in Chisel, this is a limitation of Scala. Use a Wire() construct or something,

closed 2017-02-06 by ducky64