chipsalliance/chisel #466: do_abs for fixed pt

shunshou, 2017-01-28:
Currently you have
```
def do_abs(implicit sourceInfo: SourceInfo): UInt = {
    Mux(this < FixedPoint.fromBigInt(0), (FixedPoint.fromBigInt(0)-this).asUInt, this.asUInt)
  }
```

I really think do_abs should return type T. It's confusing when the absolute value of a potentially non-integer fixed point # returns a UInt as opposed to a UFix (or just Fixed).

chick, 2017-02-02:
@shunshou seems like it's return type should be a FixedPoint.  Not doing so loses information.  I don't see any reason to make it a T.

shunshou, 2017-02-02:
Sorry I mean in general for Sint too. Just make it the same type as the input.

chick, 2017-02-02:
@ducky64 @ucbjrl I think @shunshou is right, this is a minor API issue but it seems like the current implementation is incorrect.  There seem to be very few uses of these methods, I am happy to do it if you agree

ucbjrl, 2017-02-02:
I concur.

shunshou, 2017-02-02:
It's only incorrect for fixed, but it would make sense to be consistent across the different data types.

closed 2017-02-08 by chick