chipsalliance/chisel #468: exception on incorrect enum definition

sols1, 2017-01-31:
This code works:

```
  val (s_bit_idle :: s_bit_start_a :: s_bit_start_b :: s_bit_start_c :: s_bit_start_d :: s_bit_start_e :: s_bit_stop_a :: Nil) = Enum(UInt(), 7)
```

However, if I change it to:

```
  val (s_bit_idle :: s_bit_start_a :: s_bit_start_b :: s_bit_start_c :: s_bit_start_d :: s_bit_start_e :: s_bit_stop_a :: Nil) = Enum(UInt(), 6)
```

I get an exception:

```
[info] [0.003] Elaborating design...
[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:193)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:193)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:218)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:216)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:216)
	at chisel3.Driver$.elaborate(Driver.scala:193)
	at util.HasGeneratorUtilities$class.elaborate(GeneratorUtils.scala:50)
	at sifive.enterprise.generator.EnterpriseGenerator$.elaborate(Generator.scala:40)
	at util.GeneratorApp$class.circuit(GeneratorUtils.scala:82)
	at sifive.enterprise.generator.EnterpriseGenerator$.circuit$lzycompute(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator$.circuit(Generator.scala:40)
	at util.GeneratorApp$class.generateFirrtl(GeneratorUtils.scala:88)
	at sifive.enterprise.generator.EnterpriseGenerator$.generateFirrtl(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator$.delayedEndpoint$sifive$enterprise$generator$EnterpriseGenerator$1(Generator.scala:42)
	at sifive.enterprise.generator.EnterpriseGenerator$delayedInit$body.apply(Generator.scala:40)
	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
	at scala.App$class.main(App.scala:76)
	at sifive.enterprise.generator.EnterpriseGenerator$.main(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator.main(Generator.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
Caused by: scala.MatchError: List(chisel3.core.UInt@36fc4, chisel3.core.UInt@36fc5, chisel3.core.UInt@36fc6, chisel3.core.UInt@36fc7, chisel3.core.UInt@36fc8, chisel3.core.UInt@36fc9) (of class scala.collection.immutable.$colon$colon)
	at sifive.blocks.devices.i2c.I2CModule$class.$init$(I2C.scala:164)
	at sifive.blocks.devices.i2c.TLI2C$$anonfun$19$$anon$2.<init>(I2C.scala:229)
	at sifive.blocks.devices.i2c.TLI2C$$anonfun$19.apply(I2C.scala:229)
	at sifive.blocks.devices.i2c.TLI2C$$anonfun$19.apply(I2C.scala:229)
	at uncore.tilelink2.TLRegisterRouter.module$lzycompute(RegisterRouter.scala:118)
	at uncore.tilelink2.TLRegisterRouter.module(RegisterRouter.scala:118)
	at diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:55)
	at diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:55)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:55)
	at diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:52)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at diplomacy.LazyModule.instantiate(LazyModule.scala:52)
	at diplomacy.LazyModuleImp.<init>(LazyModule.scala:133)
	at rocketchip.BareTopModule.<init>(BaseTop.scala:24)
	at rocketchip.BaseTopModule.<init>(BaseTop.scala:63)
	at sifive.enterprise.everywhere.guava.GuavaRocketChipTopModule.<init>(Top.scala:110)
	at sifive.enterprise.everywhere.guava.GuavaRocketChipTop.module$lzycompute(Top.scala:87)
	at sifive.enterprise.everywhere.guava.GuavaRocketChipTop.module(Top.scala:87)
	at sifive.enterprise.everywhere.guava.GuavaTop$$anonfun$2.apply(Top.scala:152)
	at sifive.enterprise.everywhere.guava.GuavaTop$$anonfun$2.apply(Top.scala:152)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at sifive.enterprise.everywhere.guava.GuavaTop.<init>(Top.scala:152)
	at sifive.enterprise.everywhere.guava.TestHarness$$anonfun$8.apply(TestHarness.scala:61)
	at sifive.enterprise.everywhere.guava.TestHarness$$anonfun$8.apply(TestHarness.scala:61)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at sifive.enterprise.everywhere.guava.TestHarness.<init>(TestHarness.scala:61)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:193)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:193)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:218)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:216)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:216)
	at chisel3.Driver$.elaborate(Driver.scala:193)
	at util.HasGeneratorUtilities$class.elaborate(GeneratorUtils.scala:50)
	at sifive.enterprise.generator.EnterpriseGenerator$.elaborate(Generator.scala:40)
	at util.GeneratorApp$class.circuit(GeneratorUtils.scala:82)
	at sifive.enterprise.generator.EnterpriseGenerator$.circuit$lzycompute(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator$.circuit(Generator.scala:40)
	at util.GeneratorApp$class.generateFirrtl(GeneratorUtils.scala:88)
	at sifive.enterprise.generator.EnterpriseGenerator$.generateFirrtl(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator$.delayedEndpoint$sifive$enterprise$generator$EnterpriseGenerator$1(Generator.scala:42)
	at sifive.enterprise.generator.EnterpriseGenerator$delayedInit$body.apply(Generator.scala:40)
	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
	at scala.App$class.main(App.scala:76)
	at sifive.enterprise.generator.EnterpriseGenerator$.main(Generator.scala:40)
	at sifive.enterprise.generator.EnterpriseGenerator.main(Generator.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
[trace] Stack trace suppressed: run last federation/compile:runMain for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last federation/compile:runMain for the full output.
[error] (federation/compile:runMain) Nonzero exit code: 1
[error] Total time: 11 s, completed Jan 30, 2017 3:47:48 PM
```

ducky64, 2017-01-31:
The number of values `n` you're trying to create an enum out of doesn't match the number passed in to the Enum(..., `n`). Enum returns a list, and from the MatchError I would guess it's not able to unpack the return result into the declared val.

sols1, 2017-01-31:
I know how to correct the error in the Chisel code. The point is Chisel should produce helpful error message instead of throwing an exception.

ducky64, 2017-02-06:
We don't know how to detect this and generate a better error message, though if you have suggestions a PR is welcome.

We may have a better API for Enums eventually where you don't need to manually specify the number of elements.

closed 2017-02-06 by ducky64