chipsalliance/chisel #472: Chisel vals in objects can prevent deduplication

jackkoenig, 2017-01-31:
**Workaround:** Don't use `val` in objects that return Chisel types, use `def` instead

**How do I know if I have this problem?:** When you have N of some identical module that dedups down to 2 instead of 1 final module

```scala
object MyConsts {
  val amount = 1.U
}

class ShouldDedupIO extends Bundle {
  val in = Input(UInt(32.W))
  val out = Output(UInt(32.W))
}

class ShouldDedup extends Module {
  val io = IO(new ShouldDedupIO)
  io.out := io.in + MyConsts.amount
}

class DedupTop(n: Int) extends Module {
  val io = IO(new ShouldDedupIO)
  val insts = Seq.fill(n)(Module(new ShouldDedup))
  var ref = io.in
  for (inst <- insts) {
    inst.io.in := ref
    ref = inst.io.out
  }
  io.out := ref
}
```

ShouldDedup deduplicates down to 2 final Verilog modules rather than just 1. Looking at the resulting Firrtl tells us why:
```
  module ShouldDedup :
    input clock : Clock
    input reset : UInt<1>
    output io : {flip in : UInt<32>, out : UInt<32>}

    io is invalid
    node _T_7 = add(io.in, UInt<1>("h01")) @[DedupSpec.scala 69:19]
    node _T_8 = tail(_T_7, 1) @[DedupSpec.scala 69:19]
    io.out <= _T_8 @[DedupSpec.scala 69:10]

  module ShouldDedup_1 :
    input clock : Clock
    input reset : UInt<1>
    output io : {flip in : UInt<32>, out : UInt<32>}

    io is invalid
    node _T_6 = add(io.in, UInt<1>("h01")) @[DedupSpec.scala 69:19]
    node _T_7 = tail(_T_6, 1) @[DedupSpec.scala 69:19]
    io.out <= _T_7 @[DedupSpec.scala 69:10]

  module ShouldDedup_2 :
    input clock : Clock
    input reset : UInt<1>
    output io : {flip in : UInt<32>, out : UInt<32>}

    io is invalid
    node _T_6 = add(io.in, UInt<1>("h01")) @[DedupSpec.scala 69:19]
    node _T_7 = tail(_T_6, 1) @[DedupSpec.scala 69:19]
    io.out <= _T_7 @[DedupSpec.scala 69:10]

  module ShouldDedup_3 :
    input clock : Clock
    input reset : UInt<1>
    output io : {flip in : UInt<32>, out : UInt<32>}

    io is invalid
    node _T_6 = add(io.in, UInt<1>("h01")) @[DedupSpec.scala 69:19]
    node _T_7 = tail(_T_6, 1) @[DedupSpec.scala 69:19]
    io.out <= _T_7 @[DedupSpec.scala 69:10]
```

Note that `ShouldDedup` starts at `_T_7` while `ShouldDedup_X` starts at `_T_6`. The literal being created in `MyConsts` is taking up a space in the namespace of the first instantiation of `ShouldDedup` but not in an later instantiations.

jackkoenig, 2017-01-31:
Note that literals are treated specially here: The first time the literal is accessed, it creates a literal in the current module. Subsequent references to that literal are legal even though the literal is already associated with a different module. If you were to do this with some other thing, like a Mux, Wire, or Reg, it would be an error. Perhaps this should just be an error as well.

ducky64, 2017-01-31:
Might be useful to have a literal binding? It would make sense, since literal actually turns into hardware, and it would make it more consistent when we have better chisel type / is hardware checks.

sdtwigg, 2017-01-31:
There already should be a literal binding in Binding.scala.

Regardless, the presence of the _ids list is part of the problem (along with the fact that 'everything registers' to it) since Chisel is still trying to (and failing to) name those literals. Chisel also does the same to parts of Bundles, etc.

This issue is closely related to why the T numbers jump erratically (but deterministically).

jackkoenig, 2017-04-18:
Can we change id allocation to only occur for things that need it?

sdtwigg, 2017-04-18:
Internally, I changed Chisel3 to only assign names to things that are
actually defined in commands.

This also leads to the bonus effect where T values consistently increase by
1, rather than jumping 'arbitrarily'. (The jumping is because the current
system reserves but does not use names for literals or subparts of
aggregates.)

jackkoenig, 2018-06-01:
Fixed by https://github.com/freechipsproject/firrtl/pull/766

closed 2018-06-01 by jackkoenig