chipsalliance/chisel #474: Add Chisel Line Information when generating the T_signals

jchang0, 2017-02-01:
Useful for debugging purposes when we try to figure out which signal to look at when looking at waveform which compiles the verilog.

ducky64, 2017-02-06:
Resolution: FIRRTL should propagate source location into Verilog, probably as a comment on the line of code where the variable is defined.

It seems T_ variables have source locators already associated with them in FIRRTL. Can you submit example Chisel code, FIRRTL, and Verilog that is problematic?

jchang0, 2017-02-07:
Hi,
  I'm not sure if what you are saying is that such feature already exist?  Or are you proposing a solution?  (Or is this a new feature that I haven't pulled in yet?)  I ultimately would like them them in the verilog file since it's for debugging purposes.  It's probably more helpful to have it located at the logic part of the verilog part rather than at the declarations since when I'm tracing it through verilog tools, that's where I'm going to be looking at:

I can see it have the source information in firrtl:

circuit TLCacheFormalTestBackToBack : 
  module DChannelAssocTracker : 
    input clock : Clock
    input reset : UInt<1>
    output io : 
...
...
...
   mngr_ID_mask[0] <= mngr_ID_addr_next[0] @[TileLinkScoreboard.scala 64:25]
    node _T_218 = and(io.bundle.a.ready, io.bundle.a.valid) @[Bundles.scala 207:36]
    node _T_219 = eq(io.bundle.a.bits.source, valid_IDs[0]) @[TileLinkScoreboard.scala 67:55]
...
...
 for example.

But I'm not sure where to find that information in the verilog file:

module DChannelAssocTracker(
  input   clock,
  input   reset,
.....
...
...

Here is the reference to the declaration
  reg [18:0] mngr_ID_mask_0;
  reg [31:0] _GEN_11;
  wire  _T_218;
  wire  _T_219;
....
...
...

And here is the assignments:
  assign mngr_ID_mask_next_0 = _T_274;
  assign _T_218 = io_bundle_a_ready & io_bundle_a_valid;
  assign _T_219 = io_bundle_a_bits_source == valid_IDs_0;
...

Probably more convenient to have source information where the assignments are.

Thanks,
Jacob

azidar, 2017-02-22:
Hi Jacob,

Yes, so the source locator information is being propagated to FIRRTL, but it just isn't being emitted with the Verilog. So the solution is just to emit it. It will be easier to do once I've refactored the Verilog Emitter.

ducky64, 2017-04-18:
Actually a firrtl issue, ucb-bar/firrtl#432

closed 2017-04-18 by ducky64