chipsalliance/chisel #475: Allow assignment to tuple

wsong83, 2017-02-01:
I really have no idea how difficult this would be and whether there is any language gotcha.
However, I think this feature would be convenient to have.
So in Chisel, I can do:

    val (cnt, cnt_done) = Counter(ch.valid, 20)

However, I cannot do something like:

    val a_cnt = Counter(a.valid, 20)
    val b_cnt = Counter(b.valid, 40)
    val (cnt, cnt_done) = Mux(chooseA, a_cnt, b_cnt)

Mux can only used for Data but not tuple of Data?
Also I cannot do something like:

    val m_cnt = (Wire(UInt()), Wire(Bool()))
    m_cnt := a_cnt

This is because the := operator is not defined for tuple as well.

aswaterman, 2017-02-01:
See MuxT in rocket-chip. If it becomes popular, we can export it into a common library.

colinschmidt, 2017-02-01:
I actually wanted this the other day. Good to know about MuxT

wsong83, 2017-02-02:
Thanks for the pointer to MuxT. That is exactly I am looking for.
Would be better if the length of input tuples is not limited to 2.

jackkoenig, 2017-02-02:
MuxT also has Tuples of size 3: https://github.com/ucb-bar/rocket-chip/blob/18d100c2b5b50de8e05bccf7a6f9e4ae298d8f1c/src/main/scala/util/Misc.scala#L38

But yeah, this should probably be spun out into a library and Tuples beyond Tuple3 can be supported as well.

ducky64, 2017-02-06:
possibly related to #152

ducky64, 2017-04-18:
An alternative approach is to have a type of Mux that takes Tuples.
Thoughts? Whether MuxT/Mux3T or just overload Mux?

aswaterman, 2017-04-18:
As @jackkoenig mentioned, we use them in rocket-chip.  I'd favor overloading Mux; we only named them MuxT to avoid the name collision with Chisel.