chipsalliance/chisel #478: Latest chisel3 master won't run in IntelliJ

chick, 2017-02-02:
The latest version of chisel3 master causes errors of the following form to come from ```chiselTests/NamingAnnotationTest.scala``` (10 in all).  I used to see errors like this from the chisel3 macros when they first appeared as part of sourceInfo.  Some sort of sbt magic that made sure macros were built first fixed. Maybe something similar is going on here.  Running sbt from the command lines works ok for me. 
``` 
Error:(42, 7) macro annotation could not be expanded (the most common reason for that is that you need to enable the macro paradise plugin; another possibility is that you try to use macro annotation in the same compilation run that defines it)
class NamedModule extends NamedModuleTester {
```

ducky64, 2017-02-02:
Can you see if it works with the `nameallthethings` branch? Or if you get more errors there?

chick, 2017-02-02:
@ducky64 More errors.  The errors reported above seem to have gone to be replace by a lot more like this
```
/Volumes/UCB-BAR/chisel3/src/main/scala/chisel3/compatibility.scala
Error:(350, 48) enable macro paradise to expand macro annotations
    class dump extends chisel3.internal.naming.dump
```

ducky64, 2017-02-02:
That's... different...
Is this from a project that incorporates chisel in compatibility mode and skips the chisel sbt configuration?

chick, 2017-02-02:
This is opening up the chisel3 repo and trying to run a test in it

ucbjrl, 2017-02-02:
Works for me. I do see those errors in my chisel-testers project, where I had to add the macro paradise jar as an explicit Scala compiler plugin. Initially, I thought this was because the chisel3 repo was a submodule in my chisel-testers project, but it's also a submodule in my chisel3 project (which also includes firrtl). Why things seem to work in chisel3 without having to add the macros explicitly is a mystery.

I suspect this is a bug in IntelliJ's SBT compiler plug-in handling.

chick, 2017-02-07:
Had to manually add ```.ivy2//cache/org.scalamacros/paradise_2.11.7/jars/paradise_2.11.7-2.1.0.jar```  to IntelliJ
In preferences navigate to 
Build, Execution, Deployment → Compiler → Scala Compiler
The add the above file to the Compiler plugins pane

closed 2017-02-07 by chick