chipsalliance/chisel #487: FixedPoint Lit creation inconsistent with UInt, SInt?

shunshou, 2017-02-06:
```
  implicit class fromDoubleToLiteral(val double: Double) {
      def F(binaryPoint: Int): FixedPoint = FixedPoint.fromDouble(double, binaryPoint = binaryPoint)
    }
```

For SInt, you'd do -10.S(10.W), so you're passing in a Width, but for Fixedpoint, instead of passing in a BinaryPoint, you're passing in an Int. That's confusing. 

Also, for FixedPoint, you need to specify a binaryPoint but not a width. It'd be nice if you had shorthand  for specifying a width using the .F notation too.

chick, 2017-02-08:
@shunshou I completely agree, I was noticing this when I was working on the abs function.

shunshou, 2017-02-08:
I think you'd be able to do a much better job vetting out all of these little API inconsistencies if someone bothered providing a handful of "best practices" for as many ops/Chisel-related utilities, etc. as possible (like what I'm trying to do with dsptools). The cheatsheet does not provide sufficient coverage. When I flipped through the Wiki, different examples also had different (legal) ways to create Lits, specify types, etc. In some ways, that undermines the usefulness of all this "documentation."

chick, 2017-02-08:
@shunshou It's a period of high flux, which probably argues for putting most of our documentation effort into the javadocs because the doc source is right there with the code.

shunshou, 2017-02-10:
I think people should take a survey to see if javadocs is really that great... esp. for beginners.

edwardcwang, 2017-02-10:
Doesn't the Javadoc end up on https://chisel.eecs.berkeley.edu/api/ anyway?

shunshou, 2017-02-10:
It exists for people who want to comb through all functions that exist in Chisel w/o having to read the code... but that kind of already assumes you know the typical things that you're looking for or that you're more of an "advanced" user. If you're just starting to code for the first time, you really *don't* want to be looking through JavaDocs. You want to be looking at the equivalent of https://twitter.github.io/scala_school/ for Chisel... The Wiki does that, but even then, the way it prescribes to do things is not always consistent. 

You're not going to easily capture the fact that you need to wrap Wire's with Wire() from staring at Javadoc, and if people keep asking the same question or struggling over the same thing, it means you're not doing documentation right.

shunshou, 2017-02-10:
More examples:

```
final macro def
>>(that: BigInt): Bits
 Permalink
Shift right operation
```

If the Javadoc was supposed to be "good", then it's important to let the user know also how that affects the output i.e. the width shrinks by the amount shifted, etc. Esp for FixedPoint, when things aren't immediately obvious -- are you doing a shrict arithmetic shift right or are you moving the decimal point while keeping the same # of significant bits?

Or

```
def
do_!=(that: FixedPoint)(implicit sourceInfo: SourceInfo): Bool
```

Should that even be public?!

```
def
widthKnown: Boolean
final def
widthOption: Option[Int]
```

```
final def
isWidthKnown: Boolean
 Permalink
Returns whether the width is currently known.
```

Which do you use? isWidthKnown and widthKnown seem entirely redundant? I use one version somewhere, someone else does a fixedNum.binaryPoint match { case KnownBinaryPoint(bp) => blah blah }

Of course, some of this is religious -- as in all ways are legal, but I think you should endorse one method over another for consistency... and that requires writing a how-to that's more detailed than the Cheatsheet and vetted for consistency in itself.

ben-k, 2017-02-10:
I, for one, was very excited when I learned that https://chisel.eecs.berkeley.edu/api/ existed.  Sure beats grepping through the repo!

I imagine that once 3.0.0 is released and the API is frozen, more robust wiki/cheatsheet documentation will make more sense.

closed 2017-02-17 by shunshou