chipsalliance/chisel #488: Cheatsheet: no Wire() declaration?

mwachs5, 2017-02-07:
It was suggested I file issues about the cheatsheet here.

```
val x = UInt() Allocate s as wire of type UInt()
```

First there is a typo, `x` and `a` don't match.

Second, don't you have to say 
```
val x = Wire(UInt())
```

This is, quite frankly, the thing that screwed me up the most when first writing chisel. Is this no longer true?

aswaterman, 2017-02-07:
You are correct, UInt() is only a type declaration, not a wire.

mwachs5, 2017-02-07:
Similarly in the section about Bundles & vecs, it has examples like:

```
val my_bundle = new MyBundle()
```

Should that example also be 
```
val my_bundle = Wire(new MyBundle())
```
I guess it depends on the context (is it inside a Bundle or not)?

ben-k, 2017-02-08:
Strongly second this - someone knowledgeable should take the time and make sure the cheat sheet is right here.

The mandated use of `Wire()` and friends was the most confusing thing about Chisel3 for me as well, and the bugs that result if you get it wrong can be extremely unintuitive.

mwachs5, 2017-02-24:
Ok I just realized that this is also not really even shown on the Wiki, I couldn't find any examples of Wire() on the Wiki except this one:  https://github.com/ucb-bar/chisel3/wiki/Bundles%20and%20Vecs, and actually the first example seems to also be lacking the Wire() wrap around `val x = new myFloat`.

I think you need to add something to the "Basic Users Guide to Chisel-> DataTypes" when defining all the datatypes about how to actually assign them to something.

This came up because I realized I was (still) lacking a basic understanding of something. Let's say I have a boolean that I'm going to reassign later.

```
val foo = Wire(Bool())
foo := true.B
when (complicatedStuff) {
 foo : false.B
}
```

Pretty sure the above is good. But is the below legal/equivalent, or is the inside of the Wire(...) only for figuring out the type?

```
val foo = Wire(true.B)
when (complicatedStuff) {
 foo : false.B
}
```

Finally, pretty sure the below is NOT allowed (no Wire()), but again checking my assumptions because I don't see the examples of this:

```
val foo = true.B
when (complicatedStuff) {
 foo : false.B
}
```

ucbjrl, 2017-02-24:
Chisel has types (UInt), data values (1.U), and hardware (Reg(UInt)). The
only things that can be connected are hardware and data values. Raw types,
by themselves, don't have a concrete hardware implementation, and therefore
can't participate (directly) in hardware connections. Their sole purpose is
to define the interpretation of signals that do have a concrete hardware
implementation.

If I want to define a place-holder that will participate in hardware
connections, and all I know about it is the type of signals that will
eventually participate in the connections, I define a Wire() with that
specific raw type.

I don't need to wrap literals (0.U or false.B) in a Wire() since these are
data values, and have a concrete hardware implementation.

aswaterman, 2017-02-24:
@mwachs5 The first arg to the Wire constructor is a type, not a value, so ```Wire(true.B)``` will not do what you hope.  It means, "declare a Wire that has the same type as ```true.B```", i.e., it is equivalent to writing ```Wire(Bool())```.  So the second example should use ```Wire(init=true.B)``` instead.

I believe some of this is will change prior to 3.0 release, so someone familiar with the new API (probably @jackkoenig or @ducky64) will have to audit these examples.

mwachs5, 2017-02-24:
Ok, I was digging into the code and did realize there was an init= which somehow I had not realized until now. So the best for me would be to do:

```
val foo = Wire(init = true.B)
when (complicatedStuff) {
 foo := false.B
}
```

Is there a plan to move Wire (init = ...) to WireInit similar to #455?

mwachs5, 2017-02-24:
@ucbjrl thanks for that explanation, this is something that I have figured out through confusing compile errors/behavior and have struggled to explain to new Chisel learners as well. It should really be in the Wiki with examples, as there is no such clear explanation , and there are many places as discussed on this issue where it is not correctly shown, which leads to confusion.

jackkoenig, 2017-02-24:
The working plan is to make writing `Wire(true.B)` illegal but still allow users to create values like `Wires` from the types of other values by some explicit cast.

edwardcwang, 2017-02-25:
Seems a bit inconsistent to forbid Wire(true.B) but still allow Wire(14.U)?

ducky64, 2017-02-25:
Wires of any bound ("synthesizable") types will be illegal, though you can "cast" a bound type to a data type (example, `Wire(14.U.getType())` would be legal, but nonsensical)

The init constructor is a different story though.

mwachs5, 2017-02-25:
Do you mean you plan WireInit(true.B)?

ducky64, 2017-02-25:
Yeah, probably something like that. We haven't gotten around to formally discussing Wire refactor semantics though.

aswaterman, 2017-02-25:
Wire and Reg should be symmetric, and so our discussions Wednesday about Reg should apply to Wire.

edwardcwang, 2017-04-04:
https://github.com/ucb-bar/chisel3/wiki/Combinational-Circuits#wires

mwachs5, 2017-04-04:
The example is nice, but I think you should show a more useful example.

The place I find myself using this a lot is something like:
```
val valReg = RegInit (new ValFieldsBundle().fromBits(0.U))
val valNext = Wire(init = valReg)
valReg := valNext

...
when (someCondition) {
  valNext.fieldA = true.B
}

```

mwachs5, 2017-04-04:
what I mean is, the point of declaring a Wire is that you are going to use it on the RHS of some equation, then perhaps later assign it to a different value and you expect that RHS to get the 'later' value. This isn't really clear from the example.

edwardcwang, 2017-11-08:
https://chisel.eecs.berkeley.edu/doc/chisel-cheatsheet3.pdf

ucbjrl, 2018-02-26:
Resolved (at least for the moment) by ucb-bar/chisel-doc#12.

closed 2018-02-26 by ucbjrl