chipsalliance/chisel #492: remove import chisel3.core.ExplicitCompileOptions.NotStrict

ucbjrl, 2017-02-07:
There is a:
```
// TODO: remove this once we have CompileOptions threaded through the macro system.
import chisel3.core.ExplicitCompileOptions.NotStrict
```
in 7 chisel files.
Is it time?

I'd like to replace the following in QueueIO:
```
  val enq = DeqIO(gen)
  val deq = EnqIO(gen))
```
with
```
  val enq = Flipped(EnqIO(gen))
  val deq = Flipped(DeqIO(gen))
```
to help reduce some of the confusion, but the NotStrict compile options introduce a connection error when cloneChiselType() fails to preserver directionality (due to 09282f3).

ducky64, 2017-02-07:
If it works, and after we merge #476 (to make sure that when you remove it, it's not just materializing Strict)

jackkoenig, 2017-02-12:
Fixing #495 might be required for this since people will likely use utils but pass bundles defined in Chisel._

commit 4a6396ca (2017-04-25): Remove explicit import of NotStrict - fixes #492 (#494)

* Remove explicit import of NotStrict - fixes #492

* Provide macro for MemBase.apply().

* Provide macro for MemBase.apply().
Since a macro cannot override an abstract method, provide a concrete
apply method n VecLike() that we can override with a macro.

* Remove concrete apply() in VecLike.
Since MemBase no longer extends the trait VecLike, we do not require a concrete method to which we can apply a macro to extract the appropriate CompileOptions.
commit 7449fdc9 (2017-04-26): Dropimportnotstrict492 - More updates to get things through rocket-chip. (#592)

* Remove explicit import of NotStrict - fixes #492

* Provide macro for MemBase.apply().

* Provide macro for MemBase.apply().
Since a macro cannot override an abstract method, provide a concrete
apply method n VecLike() that we can override with a macro.

* Remove concrete apply() in VecLike.
Since MemBase no longer extends the trait VecLike, we do not require a concrete method to which we can apply a macro to extract the appropriate CompileOptions.

* Add missing implicit compileOptions to do_pad() and do_zext() [... cut]
commit 293935c0 (2017-05-08): Dropimportnotstrict492 - More updates to get things through rocket-chip. (#592)

* Remove explicit import of NotStrict - fixes #492

* Provide macro for MemBase.apply().

* Provide macro for MemBase.apply().
Since a macro cannot override an abstract method, provide a concrete
apply method n VecLike() that we can override with a macro.

* Remove concrete apply() in VecLike.
Since MemBase no longer extends the trait VecLike, we do not require a concrete method to which we can apply a macro to extract the appropriate CompileOptions.

* Add missing implicit compileOptions to do_pad() and do_zext() [... cut]
commit ff5086a4 (2017-05-08): Remove explicit import of NotStrict - fixes #492 (#494)

* Remove explicit import of NotStrict - fixes #492

* Provide macro for MemBase.apply().

* Provide macro for MemBase.apply().
Since a macro cannot override an abstract method, provide a concrete
apply method n VecLike() that we can override with a macro.

* Remove concrete apply() in VecLike.
Since MemBase no longer extends the trait VecLike, we do not require a concrete method to which we can apply a macro to extract the appropriate CompileOptions.
closed 2017-04-25 by ucbjrl