chipsalliance/chisel #496: @chiselName doesn't work on Anonymous Modules

jackkoenig, 2017-02-08:
```scala
package chiselTests

import chisel3._
import chisel3.experimental.chiselName

object Broken extends App {
  @chiselName
  def doStuff(): Unit =
    chisel3.Driver.execute(args, () => new Module {
      val io = IO(new Bundle {})
    })
  doStuff()
}
```

Gives the error:
```
[error] (run-main-1) java.util.NoSuchElementException: head of empty list
java.util.NoSuchElementException: head of empty list
        at scala.collection.immutable.Nil$.head(List.scala:420)
        at scala.collection.immutable.Nil$.head(List.scala:417)
        at scala.collection.mutable.Stack.top(Stack.scala:132)
        at chisel3.internal.naming.NamingStack.pop_return_context(Namer.scala:133)
        at chiselTests.Broken$.doStuff(Broken.scala:10)
        at chiselTests.Broken$.delayedEndpoint$chiselTests$Broken$1(Broken.scala:15)
        at chiselTests.Broken$delayedInit$body.apply(Broken.scala:9)
        at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
        at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
        at scala.App$$anonfun$main$1.apply(App.scala:76)
        at scala.App$$anonfun$main$1.apply(App.scala:76)
        at scala.collection.immutable.List.foreach(List.scala:381)
        at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
        at scala.App$class.main(App.scala:76)
```

I mainly bring this up because it is nice in the chiselTests to use anonymous Modules to stop filling up the chiselTests namespace, but it's not a big deal.

ducky64, 2017-02-08:
chiselName is not to be used outside the Chisel environment (and the top-level Driver call is outside the Chisel environment).

That being said, I'm not sure how to make this apply to anonymous modules, since it's not possible to annotate individual statements, and it's also not correct to annotate a non-Chisel method (which is also why it crashes).

jackkoenig, 2017-02-08:
Fair enough, would it be possible to error saying that it can't be called outside of Chisel?

ducky64, 2017-02-08:
So the unintuitive error is the result of a deeper problem, that Builder will return a new DynamicContext even when not in a Chisel environment:

(in Builder.scala)
```scala
  private def dynamicContext: DynamicContext =
    dynamicContextVar.value.getOrElse(new DynamicContext)
```

Presumably it would make more sense for that to fail noisily (require out) if there is no dynamic context set up, and the call to DynamicNamingStack would fail inside Builder with the message along the lines of "not in a Chisel hardware construction environment".

That still makes it difficult for a new user to understand where the call to DynamicNamingStack is happening (since it'll be in the stack trace and it's a macro expansion). I don't think it's possible to determine during compile time what methods are inside a Chisel environment, so while the error message could be improved, it won't be perfect.

jackkoenig, 2017-02-08:
I think we should get rid of the default DynamicContext.

closed 2017-02-08 by jackkoenig