chipsalliance/chisel #499: Compile errors with firrtl master

donggyukim, 2017-02-09:
I have compile errors  with firrtl master as follows:

[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:7: object annotations is not a member of package firrtl
[error] import firrtl.annotations.{Annotation, CircuitName, ComponentName, ModuleName}
[error]               ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:18: not found: type Annotation
[error]   def toFirrtl: Annotation = {
[error]                 ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:19: not found: value CircuitName
[error]     val circuitName = CircuitName(component.pathName.split("""\.""").head)
[error]                       ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:22: not found: value Annotation
[error]         Annotation(
[error]         ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:23: not found: value ModuleName
[error]           ModuleName(m.name, circuitName), transformClass, value)
[error]           ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:25: not found: value Annotation
[error]         Annotation(
[error]         ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:26: not found: value ComponentName
[error]           ComponentName(
[error]           ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:27: not found: value ModuleName
[error]             component.instanceName, ModuleName(component.parentModName, circuitName)), transformClass, value)
[error]                                     ^
[warn] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:17: inferred existential type Option[(chisel3.internal.InstanceId, Class[_$1], String)] forSome { type _$1 <: firrtl.Transform }, which cannot be expressed by wildcards,  should be enabled
[warn] by making the implicit value scala.language.existentials visible.
[warn] This can be achieved by adding the import clause 'import scala.language.existentials'
[warn] or by setting the compiler option -language:existentials.
[warn] See the Scala docs for value scala.language.existentials for a discussion
[warn] why the feature should be explicitly enabled.
[warn] case class ChiselAnnotation(component: InstanceId, transformClass: Class[_ <: Transform], value: String) {
[warn]            ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/internal/firrtl/IR.scala:10: object annotations is not a member of package firrtl
[error] import _root_.firrtl.annotations.Annotation
[error]                      ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/internal/firrtl/IR.scala:278: not found: type Annotation
[error] case class Circuit(name: String, components: Seq[Component], annotations: Seq[Annotation] = Seq.empty)
[error]                                                                               ^
[warn] one warning found
[error] 10 errors found
[error] (chiselFrontend/compile:compileIncremental) Compilation failed
[error] Total time: 50 s, completed Feb 8, 2017 10:12:00 PM
> testOnly a
[info] Compiling 3 Scala sources to /scratch/dgkim/riscv-mini/chisel/coreMacros/target/scala-2.11/classes...
[info] Compiling 25 Scala sources to /scratch/dgkim/riscv-mini/chisel/chiselFrontend/target/scala-2.11/classes...
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:7: object annotations is not a member of package firrtl
[error] import firrtl.annotations.{Annotation, CircuitName, ComponentName, ModuleName}
[error]               ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:18: not found: type Annotation
[error]   def toFirrtl: Annotation = {
[error]                 ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:19: not found: value CircuitName
[error]     val circuitName = CircuitName(component.pathName.split("""\.""").head)
[error]                       ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:22: not found: value Annotation
[error]         Annotation(
[error]         ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:23: not found: value ModuleName
[error]           ModuleName(m.name, circuitName), transformClass, value)
[error]           ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:25: not found: value Annotation
[error]         Annotation(
[error]         ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:26: not found: value ComponentName
[error]           ComponentName(
[error]           ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:27: not found: value ModuleName
[error]             component.instanceName, ModuleName(component.parentModName, circuitName)), transformClass, value)
[error]                                     ^
[warn] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala:17: inferred existential type Option[(chisel3.internal.InstanceId, Class[_$1], String)] forSome { type _$1 <: firrtl.Transform }, which cannot be expressed by wildcards,  should be enabled
[warn] by making the implicit value scala.language.existentials visible.
[warn] This can be achieved by adding the import clause 'import scala.language.existentials'
[warn] or by setting the compiler option -language:existentials.
[warn] See the Scala docs for value scala.language.existentials for a discussion
[warn] why the feature should be explicitly enabled.
[warn] case class ChiselAnnotation(component: InstanceId, transformClass: Class[_ <: Transform], value: String) {
[warn]            ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/internal/firrtl/IR.scala:10: object annotations is not a member of package firrtl
[error] import _root_.firrtl.annotations.Annotation
[error]                      ^
[error] /scratch/dgkim/riscv-mini/chisel/chiselFrontend/src/main/scala/chisel3/internal/firrtl/IR.scala:279: not found: type Annotation
[error] case class Circuit(name: String, components: Seq[Component], annotations: Seq[Annotation] = Seq.empty)
[error]                                                                               ^
[warn] one warning found
[error] 10 errors found
[error] (chiselFrontend/compile:compileIncremental) Compilation failed
[error] Total time: 12 s, completed Feb 8, 2017 10:13:17 PM

jackkoenig, 2017-02-09:
Have you tried a publish-local in firrtl?

donggyukim, 2017-02-09:
I believe it should work without publish-local?

jackkoenig, 2017-02-09:
It should, but we're still in pre-3.0.0 SNAPSHOT phase and things aren't always up to date. @ucbjrl we should probably publish a new snapshot.

donggyukim, 2017-02-09:
We should never force chisel users to run publish-local. Otherwise, all projects having chisel as a submodule will be messed up.(This is why I stop using chisel testers)  I'm so scared that this can show up again even after we publish a new snapshot. It's also very unfortunate chisel annotations strongly depend on firrtl annotations.

ucbjrl, 2017-02-09:
Projects using chisel as a git submodule (like rocket-chip) specifically
include the commit they know works with their other modules. They are in
complete control of the version of chisel (and dependent repos) they use.
This has always been true.

If you use the master branch of chisel, you have to use the master branch
of all dependent repos and you have to publish-local each of those repos if
you update them. This has always been true. We do not publish the master
branches. They change much to frequently.

If you use the release branch of chisel, you have to use the release branch
of all dependent repos. In this case, these are published for you (by us),
and are guaranteed to be in sync. Currently, we are still publishing
SNAPSHOTs. Once we have an API we are committed to supporting, we will
publish tagged releases, which will not change once they've been published.

If you mix master and release branches, you're on your own.

jackkoenig, 2017-02-09:
Look at the [rocket-chip Makefrag](https://github.com/ucb-bar/rocket-chip/blob/master/Makefrag#L22) if you want to use chisel3 and firrtl as submodules

donggyukim, 2017-02-10:
So, having firrtl as an unmanaged dependency is not working with chisel3? Writing makefile in this ways is too brutal. This brings chaos the projects I'm working on.

jackkoenig, 2017-02-10:
@ucbjrl might know the more specific details, but basically sbt does not allow you to override library dependencies with a local subproject. This is a known issue https://github.com/sbt/sbt/issues/2777. The best solution we have come up with is what rocket-chip currently implements.

shunshou, 2017-02-10:
> a local subproject. This is a known issue sbt/sbt#2777

Regardless of whether sbt lets you do things, having so many project
dependencies that are all kind of part of the same thing is very user
unfriendly. Especially for people just starting out... I understand (sort
of) the point of breaking Chisel into Chisel, FIRRTL, Chisel-Testers, etc.,
but it was a lot easier from the user perspective to have one unified
Chisel, like with Chisel2.

Also, is there a better way than currently exists to be able to design
something *potentially* to use with rocket-chip but not have all of the
overhead of being compliant with rocket-chip versions of Chisel? (Besides
black boxing your stuff?) This seems to be quite problematic when you need
a feature of Chisel that's in a newer version than what rocket-chip
sub-projects.

And you're pretty much cornered into using rocket-chip style Makefrags when
you really shouldn't be needing to....

jackkoenig, 2017-02-12:
New users don't have to include lots of things, they can just include chisel3 and optionally the chisel-testers if they want those too. Regardless, new uses should use the chisel-template so that they don't have to create their own build.sbt which having to figure out is generally pretty unfriendly for new people. 

As for using rocket-chip with different versions of Chisel, I'm not sure but this seems like a good thing to look in to.

azidar, 2017-02-22:
This needs to be documented, and snapshots should be publishing more regularly.

chick, 2018-12-17:
Superseded by 556

closed 2018-12-17 by chick