chipsalliance/chisel #503: Why chisel annotations should depend on firrtl annotations?

donggyukim, 2017-02-12:
The reason why #499 happened, which will bring up disasters to all chisel & firrtl dependent projects, is chisel annotations are dependent on firrtl annotation. I have no clue why chisel annotations should depend on firrtl annotations at all. In general, it makes no sense the chisel frontend should depend on the firrtl backend. We should also remember that the motivation and the beauty of chisel & firrtl are their independences.

jackkoenig, 2017-02-12:
Chisel started depending on Firrtl before annotations actually, it started (I believe) with the execution options and other code sharing (we had a lot of duplicated code), and immediately lead to a vastly simpler API of a single function call on Chisel code to get Verilog out. This is actually why I think it makes perfect sense for chisel to depend on Firrtl. Firrtl is Chisel's IR, which means Chisel *should* natively emit Firrtl as a datastructure. Verilog is also generally the goal of writing Chisel code so it makes sense that we should make the API for doing this function simple and not require including lots of dependencies.

This being said, as discussed in #499, it is currently unacceptably difficult to use Chisel and Firrtl in a development mode--it definitely should not require `publish-local` or obnoxious Makefiles. I have thus proposed https://github.com/ucb-bar/chisel3/pull/504 where firrtl becomes an explicit submodule of chisel for development reasons. Thus rocket-chip could drop the separate construction of `firrtl.jar` and instead just use `git submodule update --init --recursive` to pull in Firrtl as well.

donggyukim, 2017-02-12:
I understand the decision, but this is still worrisome because, in theory, we should be able to use chisel with not only the scala firrtl compiler, but other firrtl compilers(e.g. the stanza firrtl compiler, or even my own c++ firrtl compiler if exists). Also, what if we have the python frontend that should work with the scala firrtl compiler?

jackkoenig, 2017-02-12:
Chisel and Firrtl are important parts of a compiler toolchain and keeping them entirely separate necessitates code duplication. We have thought about moving common code from Firrtl to a 3rd library but that just results in more repositories which seems to be an unpopular idea. It's been a longterm goal to have Chisel construct Firrtl in memory rather than just emitting a string that has to be immediately parsed again. Should we move the Firrtl IR out of Firrtl and into a 3rd repo that they both depend on to achieve this goal? Or should we stick with the strange status quo of having to go from Chisel IR to String to Firrtl IR all in the same JVM?

Nothing about this dependency stops someone from using an alternative Firrtl compiler. Besides, the dependency is *only* relevant to people who want to use chisel as a submodule--those who use it as an sbt plugin (which is what people following version numbers should do) can just include the library dependency and that is it. Alternative frontends are fine too. Chisel depending on Firrtl does not stop someone from including *just* Firrtl.

colinschmidt, 2017-02-14:
Clang requires LLVM, while LLVM **can** be built without Clang. People still are able to build things that generate and consume LLVM-IR.

I thought this was worth noting since we were nominally inspired by software compilers such as LLVM.

azidar, 2017-02-22:
This dependency is ok. The separate issue is packaging, which is being discussed in #504.

> in theory, we should be able to use chisel with not only the scala firrtl compiler, but other firrtl compilers

True, but you can always still depend on the scala firrtl compiler, and generate the text format of the firrtl design, and use your other firrtl compiler.

closed 2017-02-22 by azidar