chipsalliance/chisel #506: Cheatsheet switch statement

jchang0, 2017-02-13:
By experimentation, it looks like the for the switch-is statement, the is statement must be on a different line than the closing brace of the previous is statement

e.g. this as written on the cheetsheet does not work: (it gives this compilation error message:
overloaded method value apply with alternatives:   (x: chisel3.core.UInt)chisel3.core.Bool <and>   (x: Int)chisel3.core.Bool <and>   (x: BigInt)chisel3.core.Bool  cannot be applied to (Unit))

switch(x) {
  is(value1) {
    // run if x === value1
  } is (value2) {
    // run if x === value2
  }
}

But if I put the is statement on a different line, it works:

switch(x) {
  is(value1) {
    // run if x === value1
  }
  is (value2) {
    // run if x === value2
  }
}

Not sure if it's documentation error or chisel3 issue.

ducky64, 2017-03-01:
It looks like the first is just tripping up the Scala parser.

One attempt to solve it would be allowing `is`s to chain, parsing as `is (value1) {block1} .is (value2) {block2}`. However, that ends up in the same situation as `when` and `.elsewhen`, where if you omit the explicit dot, Scala parses `block2` as an argument for `value2`.

Will update the cheatsheet, this is bad.

ducky64, 2017-03-01:
Cheatsheet source updated.
@chick or @ucbjrl, can one of you update the compiled version on the website?

closed 2017-03-01 by ducky64

chick, 2017-03-03:
@ducky64 website updated