chipsalliance/chisel #508: reverse cat

ElvisFlewOffToMars, 2017-02-14:
Verilog has useful feature some people are used to using. It's an assignment signals concatenation to another signal. 

In Verilog it look's like: `assign {a,b,c} = z; `

Chisel3 has a `Cat()` function, but it works only for one way `z := Cat(a,b,c)` and dose not work for reverse assignment. It will be nice to create any feature for this.

ducky64, 2017-02-14:
I think the recommended way to unpack bits is to use Bundles, like
```scala
class MyBundle extends Bundle {
  val a = UInt(2.W)
  val b = UInt(4.W)
  val c = UInt(3.W)
}
...
val z = ...
val model = new MyBundle
val unpacked = model.fromBits(z)  // this will be replaced with   val unpacked = z.asTypeOf(new MyBundle)   soon
unpacked.a
unpacked.b
```

If you think that reverse-cat is still better, can you provide a full code example (like above) so we can better understand the use case?

edwardcwang, 2017-02-14:
Well, it is the difference between a one-liner and 11 lines of code...

ducky64, 2017-02-14:
Presumably you have to specify the width of a, b, c somewhere, so that's another 3 lines. And you might as well use bit slicing (though perhaps automatic bit index calculation from widths is nice, though you can get that, reusably, in a Bundle definition?)

edwardcwang, 2017-02-14:
Is there no way to infer the width of a,b,c? Maybe something like val (a,b,c) = ReverseCat(z, a, b, c)?

aswaterman, 2017-02-14:
If this sort of thing happens frequently, it usually means you should refactor your code to use structured data, as with @ducky64's example.  If you keep using the same data structures, then it becomes a one-liner, val x = new StructuredData().fromBits(y).

As to the original question, rocket-chip has a Split utility that takes widths explicitly and returns tuples (e.g., val (x, y) = Split(a, 6)).

https://github.com/ucb-bar/rocket-chip/blob/master/src/main/scala/util/Misc.scala#L118

ben-k, 2017-02-14:
It's worth noting that this question has come up repeatedly before, and our response has generally been along the lines of "won't implement - use better data structures instead".

Maybe the actionable item before closing this would be to add an item to the FAQ about it.  We ought to have a really compelling explanation about why Chisel appears (at first glance) to be missing this basic Verilog functionality, given that Chisel is at a minimum supposed to be Verilog++.

jackkoenig, 2017-02-14:
I originally created the FAQ for such often philosophical questions: https://github.com/ucb-bar/chisel3/wiki/Frequently%20Asked%20Questions

Many things need to be added but this seems like a good candidate

ElvisFlewOffToMars, 2017-02-15:
Ok, I''ll try to use fromBits. Anyway I apply the usecase. I am creating the bridge from amba axi if-ce to some packet if-ce. And some time I prefer to represent AXI channel like decoupled.

```
abstract class AXIAddrWriteChannel(addrWidth: Int) extends Bundle{

  val AWADDR = Output(UInt(addrWidth.W))
  val AWPROT = Output(UInt(3.W))
  val AWVALID = Output(Bool())
  val AWREADY = Input(Bool())

  def phData: UInt
  def phWidth: Int = phData.getWidth
  def fromPhData(data : UInt)

  def slaveConnectWithDecoupled (dec: DecoupledIO[UInt]): Unit = {
    dec.bits := phData
    dec.valid := AWVALID
    AWREADY := dec.ready
  }

  def masterConnectWithDecoupled (dec: DecoupledIO[UInt]): Unit = {
    this.fromBits(dec.bits)
    AWVALID := dec.valid
    dec.ready := AWREADY
  }
}
....................................
class AXI4AddrWriteChannel(idWidth : Int, addrWidth: Int) extends AXIAddrWriteChannel(addrWidth){
  val AWID = Output(UInt(idWidth.W))
  val AWLEN = Output(UInt(8.W))
  val AWLOCK = Output(Bool())
  val AWQOS = Output(UInt(4.W))
  val AWSIZE = Output(UInt(3.W))
  val AWBURST = Output(UInt(2.W))
  val AWCACHE = Output(UInt(4.W))//Different bits name for axi3 and axi4

  def fromPhData(data: UInt) = {
    AWID := data(idWidth + addrWidth + 25, addrWidth + 26)
    AWADDR := data(addrWidth + 25,26)
    AWSIZE := data(25,23)
    AWBURST := data(22,21)
    AWPROT := data(20,17)
    AWCACHE := data(16,13)
    AWLEN := data(12,5)
    AWLOCK := data(4)
    AWQOS := data(3,0)
  }

  def phData = Cat(AWID,AWADDR,AWSIZE,AWBURST,AWPROT,AWCACHE, AWLEN, AWLOCK, AWQOS)

  override def cloneType: this.type = new AXI4AddrWriteChannel(idWidth, addrWidth).asInstanceOf[this.type]
}
```

Maybe I should just describe it as decoupled

azidar, 2017-02-22:
Will add to FAQ.

closed 2017-02-22 by azidar

edwardcwang, 2017-03-01:
https://github.com/ucb-bar/chisel3/wiki/Cookbook#how-do-i-do-a-reverse-concatenation-like-in-verilog

aswaterman, 2017-03-01:
Thanks @edwardcwang - could I suggest renaming the pattern "unpack" instead of "reverse concatenate" and adding a justification, e.g.:

Unpacking often corresponds to reinterpreting an unstructured data type as a structured data type.  Frequently, this structured type is used prolifically in the design, and has been declared as, e.g.,

<bundle declaration>

and then it is easy to unpack the data:

<fromBits>

<and then the aside about Split>

edwardcwang, 2017-03-01:
Done!

jackkoenig, 2017-03-01:
I believe we're also switching to `someBits.asTypeOf(new MyBundle)` over `(new MyBundle).fromBits(someBits)`. If `asTypeOf` is present in the most recent snapshot release, we should probably publicize that instead.