chipsalliance/chisel #512: Module Hierarchy Proposal

azidar, 2017-02-16:
Summary of proposal from #507:

Mainline Chisel 3:

- must declare ```val io = IO(...)```
- Blackboxes drop the ```io_``` prefix to all ports
- Modules keep the ```io_``` prefix to all ports

Experimental Package:

- can declare multiple io's: ```val io1 = IO(...); val io2 = IO(...)```
- Add new `ExtModule` that keeps prefix, and users declare multiple IO's to match the underlying Verilog ports.
- Modules keep prefix

Now, users can try using Record in mainline Chisel3 to address the proliferation of Bundles in RocketChip. After this PR, users could also import experimental (knowing that this feature could be removed) and play with a solution using multiple IOs.

If, after some experimentation, multiple IO's dependence on reflection becomes a bigger hassle than the associated gains, we will get rid of this feature from experimental.

Any objections or modifications to this proposal? @ducky64 @colinschmidt @mwachs5 @sdtwigg @aswaterman @chick and anyone else.

aswaterman, 2017-02-16:
I would be more comfortable if the multi-IO thing were not called Module--even though it is in the experimental package.  If it is ultimately merged, it would be preferable to name it something else, so that Module permanently maintains the guarantee that its interface is entirely contained within (implicit clock and reset and) io.

ducky64, 2017-02-23:
Resolution:
- Will name the new multi-IO module `MultiIOModule`, in experimental so we can change the name later. This has implicit clock and reset, using the same facility as regular IO(...) signals. Remove the clock and reset constructors, users should use withClock (and friends) constructs instead.
- Use ExtModule with the multi-IO format (also experimental). In general, we will not drop `io_` prefixes.
- BlackBox and Module visible behavior stays as-is.

closed 2017-04-18 by ducky64