chipsalliance/chisel #513: setBinaryPoint

shunshou, 2017-02-17:
```
def do_setBinaryPoint(that: Int)(implicit sourceInfo: SourceInfo): FixedPoint =
    binop(sourceInfo, FixedPoint(this.width, KnownBinaryPoint(that)), SetBinaryPoint, that)
```

@chick @azidar this is the problem I was mentioning earlier. If that is > this.binaryPoint, width needs to be this.width + (that - this.binaryPoint)... I'm not sure how Chisel deals with this + if this.binaryPoint is unknown? Maybe it should just return unknownWidth?

azidar, 2017-02-22:
Yes, we should return unknownWidth if this.binaryPoint is unknown.

Another thought - can we get rid of setBinaryPoint by just calling AsFixedPoint instead?

shunshou, 2017-02-22:
If AsFixedPoint gives you the right functionality? If I take some fixed point number f, can I do f.AsFixedPoint(BP = 5), say when originally f had 3 fractional bits, and have it add 2 fractional bits + increase with to original width + 2? If so, then yes. If it doesn't do what I want, then no? lol

azidar, 2017-02-22:
Oh right, this is different behavior. Lol forget I mentioned it.

shunshou, 2017-02-22:
Prob means a better name should be thought up if the meaning isn't well conveyed :p

azidar, 2017-02-22:
Can you give me a snippet of chisel3 code that exercises this bug please?

shunshou, 2017-02-22:
```
  val test = Wire(FixedPoint(10.W, 5.BP))
  val q = test.setBinaryPoint(18)
  println(q.getWidth)
```

Width should be 10+13, instead it returns 10.

ducky64, 2017-11-08:
Appears fixed,

closed 2017-11-08 by ducky64