chipsalliance/chisel #518: edu.berkeley.cs#chisel_2.11;3.1-SNAPSHOT: not found

MahdiNazemi, 2017-02-20:
Hi,

I have cloned all four repositories related to chisel3 and ran `sbt compile` and `sbt publish-local` on them. However, when I add `libraryDependencies += "edu.berkeley.cs" %% "chisel" % "3.1-SNAPSHOT"` to my build.sbt file, I get the following error during compilation:

[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: edu.berkeley.cs#chisel_2.11;3.1-SNAPSHOT: not found
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::

Is there something I am doing wrong?

Thanks,
Matt

ucbjrl, 2017-02-20:
The library dependency you've shown is a dependency on a chisel2 SNAPSHOT.
You should be using chisel3:

libraryDependencies += "edu.berkeley.cs" %% "chisel3" % "3.1-SNAPSHOT"

Check the build.sbt in chisel-testers. As much as we would have liked to,
we couldn't rename "chisel" to "chisel2" due to compatibility issues.

MahdiNazemi, 2017-02-20:
Thanks a lot for your prompt response. I change the dependency to chisel3 and the error is gone. You may need to update the README file for chisel3 to reflect this. Currently, what I had written in my original post is written in the README file.

closed 2017-02-20 by MahdiNazemi

ucbjrl, 2017-02-21:
Thanks Mahdi. I have a pull request to update the README.