chipsalliance/chisel #521: Have an options where Asserts are not filtered with reset

jchang0, 2017-02-21:
Currently, the assertions are disabled when reset is on.

This is an issue when I want the assertions itself to explicitly consider the reset behavior (or have the assertions be based on reset behavior).

Would like to have a feature where I can deviate from default behavior and not have it generate the reset filter automatically for some assertions.

aswaterman, 2017-02-21:
Can we do this by writing: withReset (Bool(false)) { /* put the assertion here */ }

jackkoenig, 2017-02-21:
I think that should work

jackkoenig, 2017-02-21:
Keep in mind that Firrtl also guards assertions with `PRINTF_COND` which is usually set by the build system to guard assertions with the top-level reset.

azidar, 2017-02-22:
Will close this issue now, but please reopen if the proposed solution is unacceptable.

closed 2017-02-22 by azidar