chipsalliance/chisel #522: Revisit Vec wiring once the directionality / vec / binding refactors happen.

ucbjrl, 2017-02-22:
Instances of:
```
Vec.fill(n)(Module(new FullAdder()).io)
```
have to be replaced (if possible) with:
```
Array.fill(n)(Module(new FullAdder()).io)
```
since Vec wiring is now mono-connect, due to lack of `direction` on cloned `Vec` elements.

See 375e2b6a0a456c55298d82837d28986de6211ebc

sdtwigg, 2017-02-22:
I strongly doubt the mutability of the Array is actually desired so should advocate using a more appropriate Scala data structure.

Since random-indexing is highly likely, amusingly Vector is probably the best choice. To avoid confusion of Vec vs. Vector can use IndexedSeq, which is basically equivalent.

jackkoenig, 2017-02-22:
Here's a branch with a test that now fails but used to succeed:

https://github.com/ucb-bar/chisel3/tree/vec-of-modules

ducky64, 2017-03-02:
Ok, traced through this.

The issue seems to be that Vecs create an internal Wire, but Wire bindings are undirectioned, so the BiConnect operation gets really confused.

commit a6a54410 (2017-04-03): Use input element to decide if Vec of values has direction

Using the sample_element of the created wire is incorrect because Wires have no
direction so the Wire constructed for a Vec of Module IO was constructed
incorrectly. Fixes #569 and resolves #522.
PR #570 (2017-04-04): Use input element to decide if Vec of values has direction
Using the sample_element of the created wire is incorrect because Wires have no
direction so the Wire constructed for a Vec of Module IO was constructed
incorrectly. Fixes #569 and resolves #522.

Once I figured out that this was the problem, I was confused how you could index into a wire that was of type Vec of Module io and still correctly bulk connect to the directioned inputs. It turns out that this works by using the direction of the other side of the bulk connect as a hint. This has the following two nasty consequences that shoul [... cut]
commit 00796dfc (2017-04-04): Use input element to decide if Vec of values has direction (#570)

Using the sample_element of the created wire is incorrect because Wires have no
direction so the Wire constructed for a Vec of Module IO was constructed
incorrectly. Fixes #569 and resolves #522.
closed 2017-04-04 by jackkoenig