chipsalliance/chisel #524: Assertion during evaluation: assertion failed

MahdiNazemi, 2017-02-23:
I'm trying to write a minimally working floating point adder in Chisel3.
When I compile my source code for unit testing, sometimes I pass the tests and sometimes I get the following error (for the same input vector).
I don't think the problem is with the design since sometimes the tests are passed successfully. However, I can't field the issue because the error message isn't very clear for me.
 
`Assertion during evaluation: assertion failed: ERROR:11134588.PU<24> >> 139.PU<8> 139 must be < 24  @[FPAdd.scala 55:56]
Expression Evaluation stack
   0 io_out:Output -> _T_252
   1 _T_252:Node -> cat(_T_251, res_mantissa)
   2 _T_251:Node -> cat(ov_sign, res_exponent)
   3 ov_sign:Wire -> _GEN_23
   4 _GEN_23:Node -> mux(_T_82, sign, _GEN_20)
   5 _T_82:Node -> eq(_T_68, UInt<1>("h0"))
   6 _T_68:Node -> eq(_T_66, UInt<1>("h1"))
   7 _T_66:Node -> bits(xy_mantissa, 24, 24)
   8 xy_mantissa:Node -> mux(is_sub, sub, add)
   9 add:Node -> bits(_T_59, 24, 0)
  10 _T_59:Node -> add(x_mantissa, y_mantissa)
  11 y_mantissa:Wire -> _GEN_11
  12 _GEN_11:Node -> mux(_T_49, _T_54, _GEN_9)
  13 _GEN_9:Node -> validif(a_larger, _T_47)
  14 _T_47:Node -> dshr(_T_46, shift_amount)`

MahdiNazemi, 2017-02-23:
Here is some more information. It looks like the problem is related to FIRRTL:

[info]   java.lang.AssertionError: assertion failed: ERROR:16637314.PU<24> >> 91.PU<8> 91 must be < 24
[info]   at scala.Predef$.assert(Predef.scala:165)
[info]   at firrtl_interpreter.Concrete$class.$greater$greater(Concrete.scala:155)
[info]   at firrtl_interpreter.ConcreteUInt.$greater$greater(Concrete.scala:330)
[info]   at firrtl_interpreter.LoFirrtlExpressionEvaluator.dynamicBitOps(LoFirrtlExpressionEvaluator.scala:225)
[info]   at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:349)
[info]   at firrtl_interpreter.LoFirrtlExpressionEvaluator$$anonfun$5.apply(LoFirrtlExpressionEvaluator.scala:453)
[info]   at firrtl_interpreter.LoFirrtlExpressionEvaluator$$anonfun$5.apply(LoFirrtlExpressionEvaluator.scala:445)
[info]   at firrtl_interpreter.Timer.apply(Timer.scala:40)
[info]   at firrtl_interpreter.LoFirrtlExpressionEvaluator.firrtl_interpreter$LoFirrtlExpressionEvaluator$$resolveDependency(LoFirrtlExpressionEvaluator.scala:444)
[info]   at firrtl_interpreter.LoFirrtlExpressionEvaluator.getValue(LoFirrtlExpressionEvaluator.scala:81)

jackkoenig, 2017-02-23:
If you pass the tests sometimes, but not always, it sounds to me like the kind of nondeterminism that comes from randomization. Randomization occurs when something is not initialized.

I'm not 100% sure about how the interpreter handles this, but `13 _GEN_9:Node -> validif(a_larger, _T_47)` says that if `a_larger` is true, then `_GEN_9` will get the value from `_T_47`. However, if `a_larger` is false *I believe* that the interpreter will set `_GEN_9` to a random value.

Do you have code I can see to reproduce?

MahdiNazemi, 2017-02-23:
The error is sometimes this:

   0 io_out:Output -> _T_251
   1 _T_251:Node -> cat(_T_250, res_mantissa)
   2 _T_250:Node -> cat(ov_sign, res_exponent)
   3 ov_sign:Wire -> _GEN_23
   4 _GEN_23:Node -> mux(_T_81, sign, _GEN_20)
   5 _T_81:Node -> eq(_T_67, UInt<1>("h0"))
   6 _T_67:Node -> eq(_T_65, UInt<1>("h1"))
   7 _T_65:Node -> bits(xy_mantissa, 24, 24)
   8 xy_mantissa:Node -> mux(is_sub, sub, add)
   9 add:Node -> bits(_T_58, 24, 0)
  10 _T_58:Node -> add(x_mantissa, y_mantissa)
  11 y_mantissa:Wire -> _GEN_11
  12 _GEN_11:Node -> mux(_T_48, _T_53, _GEN_9)
  13 _T_53:Node -> dshr(_T_52, shift_amount)

which doesn't have the validif statement anymore.

I'll try to clean up my code and upload it here, but I guess it's gonna take some time.

MahdiNazemi, 2017-02-23:
Thanks a lot Chick. I modified the code for dynamic right shift and the issue is resolved.

MahdiNazemi, 2017-03-15:
I had addressed this issue using a mux and by outputting zero when the shift amount was greater than the number of bits in the original variable. However, this solution doesn't work anymore. Do you have any suggestions or workarounds for suppressing this error?

chick, 2017-04-18:
The interpreter problem uncovered here was fixed under https://github.com/ucb-bar/firrtl-interpreter/pull/75

closed 2017-04-18 by chick