chipsalliance/chisel #526: "Default Block" / more than one match per line for switch statement

jchang0, 2017-02-23:
Two separate feature requests

1. Have a default block available for switch statements (where if none of the is block match, it will execute the default block)
e.g.
switch(value) {
is(1) {
// do something
}
is(2) {
// do something
}
default {
  error := true.B
  assert(0)
}
}

2. Allow more than one matched value for each is block match.
e.g. (the first two cases are more useful, the subsequent examples have potential usages, but I haven't encountered it yet)

switch(value) {
is(1) {
// do something
}
is(2, 4) {
// do something
}
// we might need to figure out what it means if the user duplicate a value in this case (perhaps execute both is blocks?)  Or just error out and disallow it if it overlaps.
is(4, 6) {
}
// or maybe even more fancy, but can no longer check for non-overlapping matches if that's a requirement
is(>5) {
}
// Or even generalizing it by allowing a predicate function (Maybe this and the previous is too much, since if it gets that fancy, should probably just use when statement, and feature not in verilog anyway)
is( (v: Int):Bool => { v<7 }) {
}
}

mwachs5, 2017-02-23:
What is the benefit of switch over when/elsewhen/otherwise ? I feel like switch is never used...

sdtwigg, 2017-02-23:
On 2, you can actually already (most of) this. The is function takes a varargs. It will run both blocks if you repeat a value in two blocks (although unclear whether this is guaranteed behavior or just a side effect of the implementation).

@mwachs5 switch is currently immediately (at frontend) transcribed into "when, when, when" so the only benefit is in potential Chisel readability.

jackkoenig, 2017-11-08:
I think there's more to this that we should consider that is an API modification. Because switch emits when, when, when, instead of when, elsewhen, otherwise it leaves coverage holes in the Verilog. This makes it a terrible thing to use for state machines which is what it's really intended for. We probably should change it (but should also check what Verilog case statements do). This would be an API backwards incompatible change though.

ducky64, 2017-12-13:
Resolution: do both by 3.1

azidar, 2019-10-07:
#1198 is related. Conclusion: is a Challenge Issue.

juliusbaxter, 2020-01-31:
Just a ping that this would be useful 👍 

I notice it didn't make it into the latest Chisel release as per:
https://www.chisel-lang.org/api/latest/chisel3/util/switch$.html

Thanks!

samadpls, 2022-09-05:
> Two separate feature requests
> 
> 1. Have a default block available for switch statements (where if none of the is block match, it will execute the default block)
>    e.g.
> 
> ```scala
> switch(value) {
>   is(1) {
>   // do something
>   }
>   is(2) {
>   // do something
>   }
>   default {
>     error := true.B
>     assert(0)
>   }
> }
> ```
> 
> 2. Allow more than one matched value for each is block match.
>    e.g. (the first two cases are more useful, the subsequent examples have potential usages, but I haven't encountered it yet)
> 
> ```scala
> switch(value) {
>   is(1) {
>   // do something
>   }
>   is(2, 4) {
>   // do something
>   }
>   // we might need to figure out what it means if the user duplicate a value in this case (perhaps 
>   execute both is blocks?)  Or just error out and disallow it if it overlaps.
>   is(4, 6) {
>   }
>   // or maybe even more fancy, but can no longer check for non-overlapping matches if that's a requirement
>   is(>5) {
>   }
>   // Or even generalizing it by allowing a predicate function (Maybe this and the previous is too much, since if it gets that fancy, should probably just use when statement, and feature not in verilog anyway)
>   is( (v: Int):Bool => { v<7 }) {
>   }
> }
> ```

thankyou so much for this, I have been searching for this for so long.