chipsalliance/chisel #527: Have to assign IO Vec to intermediate Wire in elsewhen

mwachs5, 2017-02-23:
The below is from WIP branch on rocket-chip (https://github.com/ucb-bar/rocket-chip/blob/debug_0.13/src/main/scala/uncore/devices/Debug.scala#L319) . I tried to isolate it more but couldn't get a repeatable case. This is not blocking me at all, and there may be other things going on in this code that are the real problem, but logging it here in case there is something obvious going on.

I have a a Vec of Bools as part of my input,  and I want to select on a bit of it in a `when` statement:

```
trait DebugModuleBundle extends Bundle with HasDebugModuleParameters {
...
  val debugUnavail    = Vec(cfg.nComponents, Bool()).asInput
...
}

(the trait is mixed in for the IO for the module)

when (...) {... 
} .elsewhen(io.debugUnavail(DMCONTROLReg.hartsel)) {
...
  } 
```
The line above gives me a FIRRTL error shown below. If I replace it with the following, it works fine:

```
//MUST put the Wire declaration, or this has the same error
val unavailVec = Wire(Vec(cfg.nComponents, Bool()))
unavailVec := io.debugUnavail

when (...) {... 
} .elsewhen(unavailVec(DMCONTROLReg.hartsel)) {
...
  } .otherwise {
...
  }
```

Error I get:

```
error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
	at chisel3.core.Module$.do_apply(Module.scala:42)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:234)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:232)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:232)
	at chisel3.Driver$.elaborate(Driver.scala:91)
Caused by: chisel3.core.Binding$BindingException: 'other' (chisel3.core.Bool@287b0): Not bound to synthesizable node, currently only Type description
	at chisel3.core.Binding$.checkSynthesizable(Binding.scala:186)
	at chisel3.core.Bits.binop(Bits.scala:163)
	at chisel3.core.Bool.do_$amp(Bits.scala:724)
	at chisel3.core.Bool.do_$amp$amp(Bits.scala:744)
	at chisel3.core.WhenContext.elsewhen(When.scala:47)
	at uncore.devices.DebugModule$class.$init$(Debug.scala:330)
	at uncore.devices.TLDebugModule$$anonfun$42$$anon$2.<init>(Debug.scala:712)
	at uncore.devices.TLDebugModule$$anonfun$42.apply(Debug.scala:712)
	at uncore.devices.TLDebugModule$$anonfun$42.apply(Debug.scala:712)
	at uncore.tilelink2.TLRegisterRouter.module$lzycompute(RegisterRouter.scala:118)
	at uncore.tilelink2.TLRegisterRouter.module(RegisterRouter.scala:118)
	at diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:60)
	at diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:60)
	at chisel3.core.Module$.do_apply(Module.scala:42)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2081)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:740)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:738)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
```

ducky64, 2017-04-18:
May be a compile options bug, seems bindings are leeching in Chisel compatibility

ucbjrl, 2017-05-11:
The following works for me with chisel3 8baa2ab :
```
package chiselTests

import Chisel._

trait DebugModuleBundle extends Bundle {
  val nComponents = 4
  val debugUnavail    = Vec(nComponents, Bool()).asInput
}

class ElseWhenVec extends Module {
  val io = new DebugModuleBundle {
    val in = UInt(16.W).asInput
    val out = UInt(16.W).asOutput
    val trigger = Bool().asInput
  }
  when (io.in === 0.U) {
    io.out := io.in
  } .elsewhen ( io.debugUnavail(0.U) ) {
    io.out := 1.U
  }
}

class ElseWhenVecSpec extends ChiselPropSpec {
  property("ElseWhenVec should elaborate") {
    elaborate { new ElseWhenVec }
  }
}
```
At least I don't get any `BindingException`. It also works with `import chisel3._` if I wrap the `io` definition in `IO()`.

mwachs5, 2017-05-12:
Let me take another look at it. Perhaps it was confounded by other factors.

ducky64, 2017-11-08:
@mwachs5 Is this still an issue?

mwachs5, 2017-11-27:
I'll try it out today and see if it still fails.

mwachs5, 2017-11-27:
No, it doesn't fail anymore (using rocket-chip which is on chisel3 e2c5c12)

closed 2017-11-27 by ducky64