chipsalliance/chisel #529: Can no longer create Vecs of size 0 with a literal type

jchang0, 2017-02-24:
I think this is due to due to commit #480: Added vec IO tests for #104 

where the construction method for Vec(Seq) construction requires Seq to be not empty.

Actually, it would be useful for Vecs to be able to create an empty vec even with the Vec(Seq) construction.  Useful when Seq is constructed from parameter, and a 0 length sequence is a possible parameterization, and don't want to write a special case handler.

jackkoenig, 2017-04-18:
The real question is what is the type of an empty Vec? In Scala, all empty Seqs are equal despite having different types. 
eg.

```
scala> val x = Seq.empty[Int]
x: Seq[Int] = List()

scala> val y = Seq.empty[String]
y: Seq[String] = List()

scala> x == y
```

So what should we do in Chisel?

aswaterman, 2017-04-18:
I agree this is a bug.  They're equal in Scala because the type doesn't matter in that case.  Seems like the same should be true of Vec.

As for the implementation, it should be made possible to pass ```gen = null.asInstanceOf[T]``` when size==0.

closed 2017-11-08 by ducky64