chipsalliance/chisel #535: A simple proposal for better generated names

ben-k, 2017-02-27:
Although Chisel3 naming is definitely improving, I still find myself having to grep through the generated Firrtl for a wire/reg/module/etc. named `_T*` every so often.

This process would be a bit more user friendly if the Chisel-generated names were distinguished by type.  So wires could be named `_W*`, registers `_R*`, modules `_M*`, and so forth.

I imagine (hope) this would not be a burdensome change, and it would definitely improve readability and ease my use case.

aswaterman, 2017-02-27:
Seems reasonable to me.  Would also have the effect of making temp naming for registers and memories ever-so-slightly more stable, since the majority of temps are wires.

jackkoenig, 2017-02-27:
I agree with the suggestion, one thing to note:

Modules *should* already be named based on the name of the Module itself. If this is not the case, it is a bug.

aswaterman, 2017-02-27:
when I first skimmed the proposal I assumed _M was for memories... which it should be.

colinschmidt, 2017-02-27:
This specific case of a module with a _T name is a SeqMem that becomes a module during ReplSeqMem, so the _M stands for memory.

jackkoenig, 2020-10-16:
This was done in `v3.4.0`

closed 2020-10-16 by jackkoenig