chipsalliance/chisel #536: Analog type does not allow static addressing of sub-wires

jwright6323, 2017-02-27:
For example, if you have a 4-bit wide Analog bus and want to route each wire to separate blocks, you cannot do the following:

...
val myWire = Analog(4.W)
...
val bb = Module(new MyBlackBox())
attach(bb.io.myWire,io.myWire(3))
...

It looks like an apply(x: Int) method like in Bits would be sufficient.

jackkoenig, 2017-02-27:
Thanks for filing an issue to track it. Fixing this is on my TODO list but it's not quite as trivial as copying the code from Bits, the main implementation is in Firrtl. In the meantime, you can use Blackboxes.

jwright6323, 2017-02-27:
Sounds good- I have a workable solution.

shunshou, 2017-02-27:
Exactly my problem last week. Black boxes are actually a pretty nifty "hack" solution -- in some sense, actually cleaner than whatever Firrtl emits.