chipsalliance/chisel #538: Expected chirrtl output of bulk connections in the compatibility mode?

donggyukim, 2017-02-28:
I guess bulk connections were partial connections before the gsdt commit, but now it seems like they are all expanded in chisel even in the compatibility mode. Unfortunately, this breaks boom as in ucb-bar/riscv-boom#20.

jackkoenig, 2017-02-28:
Bulk connections should not be being expanded in compatibility mode cde. Can you point me to compatibility code that is having its bulk connects broken up?

donggyukim, 2017-02-28:
With 224ee39, bulk connections were expanded. I have no idea why.

jackkoenig, 2017-02-28:
I'm using master chisel3 and firrtl on rocket-chip DefaultConfig and I see bulk connects preserved as firrtl partial connects.

donggyukim, 2017-02-28:
Please refer to this:
https://github.com/ucb-bar/riscv-boom/blob/master/src/main/scala/dcacheshim.scala#L196

This is the diff:
<     inflight_load_buffer_out_uop[0] <- LoadReqSlot_4.io.out_uop @[dcacheshim.scala 322:42]
<     inflight_load_buffer_out_uop[1] <- LoadReqSlot_1_1.io.out_uop @[dcacheshim.scala 322:42]
<     inflight_load_buffer_out_uop[2] <- LoadReqSlot_2_1.io.out_uop @[dcacheshim.scala 322:42]
<     inflight_load_buffer_out_uop[3] <- LoadReqSlot_3_1.io.out_uop @[dcacheshim.scala 322:42]
---
>     inflight_load_buffer_out_uop[0].debug_events.fetch_seq <= LoadReqSlot_4.io.out_uop.debug_events.fetch_seq @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].debug_wdata <= LoadReqSlot_4.io.out_uop.debug_wdata @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].replay_if <= LoadReqSlot_4.io.out_uop.replay_if @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].xcpt_if <= LoadReqSlot_4.io.out_uop.xcpt_if @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].fp_single <= LoadReqSlot_4.io.out_uop.fp_single @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].fp_val <= LoadReqSlot_4.io.out_uop.fp_val @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].frs3_en <= LoadReqSlot_4.io.out_uop.frs3_en @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].lrs2_rtype <= LoadReqSlot_4.io.out_uop.lrs2_rtype @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].lrs1_rtype <= LoadReqSlot_4.io.out_uop.lrs1_rtype @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].dst_rtype <= LoadReqSlot_4.io.out_uop.dst_rtype @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].ldst_val <= LoadReqSlot_4.io.out_uop.ldst_val @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].lrs3 <= LoadReqSlot_4.io.out_uop.lrs3 @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].lrs2 <= LoadReqSlot_4.io.out_uop.lrs2 @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].lrs1 <= LoadReqSlot_4.io.out_uop.lrs1 @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].ldst <= LoadReqSlot_4.io.out_uop.ldst @[dcacheshim.scala 322:42]
>     inflight_load_buffer_out_uop[0].flush_on_commit <= LoadReqSlot_4.io.out_uop.flush_on_commit @[dcacheshim.scala 322:42]
..... (a lot...)

jackkoenig, 2017-02-28:
Right so I think this is the problem with `Vec` that was recently introduced and hopefully fixed in https://github.com/ucb-bar/chisel3/pull/530, can you try that branch?

donggyukim, 2017-02-28:
With #530, there an error for https://github.com/ucb-bar/riscv-boom/blob/fix-vec/src/main/scala/rob.scala#L701:

[error] /scratch/dgkim/fpga-zynq/rocket-chip/boom/src/main/scala/rob.scala:701: not enough arguments for method do_indexWhere: (implicit sourceInfo: chisel3.internal.sourceinfo.SourceInfo, implicit compileOptions: chisel3.core.CompileOptions)chisel3.core.UInt.
[error] Unspecified value parameter compileOptions.
[error]          val idx = dis_xcpts.indexWhere{i: Bool => i}

jackkoenig, 2017-02-28:
@ucbjrl Thoughts on the above error?

ucbjrl, 2017-02-28:
Yes. Sounds like it's missing an implicit. Hang on.

ucbjrl, 2017-02-28:
I've pushed a (potential) fix. I'll add a test.

donggyukim, 2017-02-28:
#530 preserves the partial connects for Vec, so I'm closing this.

closed 2017-02-28 by donggyukim