chipsalliance/chisel #539: asUInt and allies should not have parens in def with no parameters

chick, 2017-03-01:
asUInt is defined in several places as ```def asUInt()```.  Standard usage of this method
is to say something like ```val x = someSInt.asSInt```.  Style checkers do not like this inconsistent usage.  Ordinarily parens on a method like this indicated side effects, so I would propose that the API for parameter less as<X> methods be changed to not use empty parens.
This could cause some existing code to get complaints, but should not break anything

sdtwigg, 2017-03-01:
I am reasonably certain the () in the definition is needed for overloading purposes.

Unless things have changed since I last experimented...
```scala
def asUInt()
def asUInt(width: Width)
```

but this will cause issues calling the width variant
```scala
def asUInt
def asUInt(width: Width)
```

Note that this was discovered when using the extension methods on String, Int, and BigInt. Unsure at the moment if it applies to normally defined methods.

chick, 2017-03-01:
@sdtwigg  Quick test with a scala worksheet, is this what you meant?
```
class X {
  def f: Int = 7
  def f(x: Int): Int = x + 7
}

val x = new X
x.f => 7
x.f(9) => 15
```

chick, 2017-06-19:
I think this should be closed or moved back.  The fix is easy but it is API breaking.  Anyplace a dev has written 7.U() will be a compiler failure.  This occurs  a few times in the freechips repos.  There is also some confusion generated in that macros like asUInt must be declared with the empty parens despite the convention that they are parameterless calls without side-effects.  This would not seem to be on the critical path for 3.0.0

closed 2017-07-28 by chick