chipsalliance/chisel #540: Add assertion example to Wiki

mwachs5, 2017-03-02:
The Wiki has examples of how to print, but not write assertions. My assertions recently stopped compiling, perhaps because of #500 ? What is the right way to do something like:

`assert (a == b, "A (%x) is not equal to B (%x) !", a, b)`

If I do this I get errors like:

```
Caused by: java.lang.IllegalArgumentException: requirement failed: Too many arguments! More format specifier(s) expected!
    fmt = "A (%%x) is not equal to B(%%x)!"
...
```

jackkoenig, 2017-03-02:
Whoops. This is my fault. We do need examples but this should not have stopped working. I think the fix shouldn't be too bad.

jackkoenig, 2017-03-02:
Can you try https://github.com/ucb-bar/chisel3/pull/542 to see if it fixes your broken assertions?

mwachs5, 2017-03-03:
Yes it fixes my issue

commit 4cad5fc7 (2017-05-04): Fixes format strings in assertions. Fixes #540
commit cb7fcd2b (2017-12-22): Fixes format strings in assertions. Fixes #540 (#542)
closed 2017-12-22 by jackkoenig