chipsalliance/chisel #541: API to walk through module scala instances

jchang0, 2017-03-02:
I would like to be able to walk though a module hierarchy (for this particular application, the Scala instances) in order to collect information about the design (or in the future, generate monitor code to be bound later).
So an API to traverse through the module instances would be helpful.

shunshou, 2017-03-02:
Do you want this to be done in Chisel or in Firrtl? I've requested something similar to this for the purposes of auto-generating some clock constraint information at the Firrtl level. Assuming the module name is unique, @albert-magyar wrote a cutesy analysis thing that can map the module name to its absolute instance path. (If the module name is not unique, it'll get all instance paths to that module). I'm pretty sure @jackkoenig @azidar are aware of how powerful something like this can be, and really, I think this is where you can really exercise Firrtl potential...

jackkoenig, 2017-03-02:
I think there are reasons you would want it in both. In Firrtl for obvious
reasons, but also in Chisel it can be nice to be able to inspect the actual
Chisel Scala objects.

<https://github.com/ucb-bar/chisel3/issues/541#issuecomment-283784126>, or mute
the thread

jchang0, 2017-03-02:
For this particular need I'm currently encountering, in Chisel (I want to access the LazyModule parameters so that I can create a separate breakout unit level tester with the same parameters as the final instantiation).  But I've talked to jack about firrtl passes also for other purposes in the past also.

ducky64, 2017-12-13:
Related: #414

azidar, 2019-10-07:
This is implemented in #1077 !

closed 2019-10-07 by azidar