chipsalliance/chisel #545: Incremental compilation

colinschmidt, 2017-03-04:
Is it possible?

sdtwigg, 2017-03-04:
What do you mean?

Do you mean running continuously re-running a chisel3 project without closing sbt so you get access to zinc for scala incremental recompilation? Or, some sort of firrtl incremental recompilation?

colinschmidt, 2017-03-04:
In an ideal world, If a small change is made to a chisel file  both less scala code needs to be run as well as only the new firrtl was emitted and then a firrtl compiler could take advantage of this when generating the verilog.

jackkoenig, 2017-03-04:
I'm not sure how to do this in Chisel, but there are a couple of ways we might be able to do this for Firrtl:

1. Chisel emits Modules to separate files (and possibly a Makefile .d file). We could then use Make to determine which files have changed and need to have Firrtl rerun on. Bonus would be parallelizing the build on a per-module basis. Not sure if something similar is possible in Sbt and not sure how we would handle inferring port widths here.
1. Firrtl adds native support for incremental compilation. I'm not sure what support sbt (or Make) might provide for this, but I believe it would need to store some state between runs that it would use to determine what to recompile.

sdtwigg, 2017-03-04:
So, regarding one and firrtl level incremental recompilation, you generally can't easily do this due to unknown input port width and whatever annotations people add (that apparently may add or remove logic in submodules). This is why in sbt it tends constantly says it is compiling a different number of files: Zinc is looking at the at the files specifically and determining the interdependencies and issuing various ones to compile in order. Similarly, this is why people advise you properly ascribe types to public members of classes: Zinc can then use the old ijar to compile dependencies while recompiling the implementation. (It has been a while since I remembered most of the things Zinc does so I will have to ask a friend for a refresher.)

Emitting separate files shouldn't technically matter (you can just splice the original file on module boundaries already). You will run into more significant issues with Chisel's dramatic renaming of module definitions in a design due to subtle changes. (Add a new Queue somewhere in top level near top of file and suddenly you are recompiling every Queue and every user of a Queue and so on.) You would need to design a system to either avoid these renames or match old names to the new names (and then rewrite user bodies).

abukharmeh, 2022-11-17:
Any updates on this one ?