chipsalliance/chisel #548: Trying to programmatically generate LUT black box (naming help???)

shunshou, 2017-03-11:
I'm trying to programmatically generate LUT black boxes (so that I won't be restricted by Chisel3's lack of ROM support), but I'm running into the problem of naming. Since I'm programmatically generating the LUTs, I don't know ahead of time what the names are. Since Chisel derives names from module classes, I'm kind of stuck? 

I could update the module name inside the black box's Verilog file, but that requires templating the name and making modifications to BlackBoxSourceHelper. Or is there some mechanism to override the way that Chisel3 names things? I can't easily find it if it does exist...

jackkoenig, 2017-03-11:
Override desiredName. Chisel uses desiredName which by default derives from
the class name.

shunshou, 2017-03-11:
What happens if the desiredName ends up being not unique? I don't want to wait until I simulate with Verilator to know that there's a black box name conflict. (Or if Chisel uniquifies the name but that new name ends up not agreeing with the black box module name).

jackkoenig, 2017-03-11:
Given that Black Boxes are usually IP blocks with set names, Chisel and
Firrtl guarantee that the final instantiated module uses the desiredName,
there is no uniquification. If you need to guarantee the names are unique
for black boxes, you have to do it manually.

shunshou, 2017-03-11:
Ok, that's fair. Thanks! Idk if this is worth documenting in the wiki or elsewhere...

closed 2017-04-18 by jackkoenig