chipsalliance/chisel #549: getResources Bug

shunshou, 2017-03-12:
I'm trying to port over some of my previous work from Chisel2 to Chisel3. I was hoping to have one project depending on both Chisel2 and Chisel3 so I could incrementally port some stuff over (not running both at the same time). I ran into a really nasty issue when trying to get Verilator tests to work where (chisel-testers) VerilatorBackend's getResources (or w/e it's called) was pulling in the sim_api.h from chisel2 instead of the sim_api.h in chisel-testers. 

If your top-level project depends on multiple sbt-projects whose src/main/resource directory contains files of the same name, it's painful to figure out how to get the precedence to be right. 

The directory structure should be src/main/resource/package_name or something to disambiguate and all of your getResource args should reflect the package_name.

I know this doesn't immediately affect 99% of users, but wasting time debugging this without understanding how Scala/sbt works can be a huge time sink, so please fix this!

ucbjrl, 2017-03-12:
This is related to ucb-bar/firrtl#398 (which is still waiting for review).

shunshou, 2017-03-14:
@ucbjrl how can we move this along so other people don't run into weird problems?

ucbjrl, 2017-03-16:
I propose we ensure resources are explicitly scoped in ucb-bar jars - there should be a top-level resource directory with the name of the project/jar - i.e., firrtl resources should live in `src/{main,test}/resources/firrtl/...`, chisel2 resources in `src/{main,test}/resources/chisel2/...` etc. We can discuss this at the next chisel meeting.

chick, 2017-04-10:
added this to chisel action items

commit 8baa2ab8 (2017-06-17): Scope resources - move them down into chisel3 directory - fixes #549 (#610)
closed 2017-05-11 by ucbjrl