chipsalliance/chisel #550: Vec of lits addressing

shunshou, 2017-03-13:
Has anyone thought about what the default should be if your Vec of Lits was not some 2^n depth?Adam says it gets the zero-addressed element right now. Chisel2 Verilog dump shows that it set "X" as a default... maybe random number is better? If nothing else, I think the behavior is good to document, but maybe setting to random would help with debugging if the user wrote bad code.

colinschmidt, 2017-03-13:
Is there any reason it shouldn't just do what non-literal Vecs do when addressed outside of depth? I thought that behavior was random but I haven't checked in a long time.

shunshou, 2017-03-13:
I haven't checked -- my reasoning for asking was to determine what I should be doing (= best practice) for my LUT black boxing stuff (esp since I have a lot of weird non power of 2 LUTs). 

If you think random is the best, then I'll stick to that. Still, it's a good idea to document the behavior or make documentation apparent w/o having to run code through.

ducky64, 2017-04-18:
We think behavior should be undefined when addressing out of range.
@aswaterman is this right?

aswaterman, 2017-04-18:
Yeah

ducky64, 2017-11-08:
Resolution: make this behaviorally consistent with the output being Invalid, and backends can either randomize or X as consistent

ducky64, 2017-12-13:
will be addressed by freechipsproject/firrtl#685

closed 2018-01-04 by ducky64