chipsalliance/chisel #555: withClockAndReset + node names

shunshou, 2017-03-20:
Not really sure how thiClockAndReset works, but it looks like if you wrap withCLockAndReset around a bunch of internal signals (so you do it once at a module level instead of individually), say

```
 withClockAndReset(io.clk, io.stateInfo.start) {
    val currStateIsDebug = Wire(Bool())
        currStateIsDebug := Mux1H((0 until states.toSeq.length).map(x => (io.currentState === x.U) ->      io.scr.debugStates(x)))
```

currStateIsDebug gets created as _T something. However, if you don't wrap it, a useful name actually gets inferred.

jackkoenig, 2017-03-20:
Right so `withClockAndReset` wraps things in a scope which manes the `val`s are no longer fields of the actual Module class so they don't get named via reflection. This is the same thing as putting a `val` in a `for` loop or in any anonymous function.

This is why @ducky64 worked so hard on the naming macros! Try out `@chiselName`! (see https://github.com/ucb-bar/chisel3/blob/master/src/test/scala/chiselTests/NamingAnnotationTest.scala#L56 for how to use)

shunshou, 2017-03-20:
Uh that doesn't really do a great job of indicating how to use it. Do you just add @chiselName in front of the module you want names to be well defined for?

ducky64, 2017-03-21:
Yeah, that's basically how it works, the rest is magic. You will also need to `import chisel3.experimental.chiselName` though.

What this macro does is that it expands all instances of `val x = ...` into a naming call, even if they're inside an inner scope that normally wouldn't be visible by reflection. There's also additional features for propagating names through functions, you can also `@chiselname` those.

shunshou, 2017-03-21:
Why would you not want to do this for everything?

ducky64, 2017-03-21:
You probably would want this on everything, though it's still experimental. Suggestions / comments / feedback are welcome too!

(if you're asking why this isn't automatically applied to all Modules, we don't quite know how to do that. It's also completely opt-in, just in case something goes horribly wrong)

shunshou, 2017-03-21:
Was just hoping that once you import chisel3.experimental.chiselName, it'd just use it everywhere...

ducky64, 2017-03-21:
As far as I know there isn't a way to do file-level macros (at least without going into really exotic things like compiler plugins). `@chiselName` uses macro paradise's annotations, which have to be applied at a function or class level.

closed 2017-03-21 by jackkoenig

shunshou, 2017-03-23:
More general question: @ducky64 is there some way to get chiselName to work for something like:
```
val writeBankSel = memBanks.io.bank.elements.map { case (bankIdx, _) =>
       bankIdx.toInt -> io.i.map { case lane => lane.loc.bank === bankIdx.toInt.U }		      
}.toMap}.toMap
```

As an alternative, I can always create a Wire consisting of nested Records... but I wish it'd be able to pick up on map names, since I use constructs similar to this a lot.

jackkoenig, 2017-03-23:
It kind of surprises me that the unapply doesn't get transformed by the compiler into an assignment to a val

ducky64, 2017-03-23:
@shunshou What names do you want to have the system automatically produce? It looks like quite the complicated example...

shunshou, 2017-03-23:
Call the 2d mapped bools writeBankSel ?

ducky64, 2017-03-23:
So I think the issue here is that the naming annotation doesn't look inside containers - if I'm understanding your code right, `writeBankSel` isn't a Chisel Data type, it's a Map[Int, Data] right?

Anyways, this is one place where the annotation behavior currently diverges from the reflection naming (which looks inside iterables). This can be fixed.

shunshou, 2017-03-23:
Yeah more like map[int, map[int, data]] stupidly but I use that a lot to make generator stuff easier.

jackkoenig, 2017-03-23:
The reflective naming looks inside collections but not collections of Tuples so currently Map is not supported. It definitely could be but I'm not sure what the right way to do it is. For example:

```scala
val myMap = Map(0 -> Reg(UInt(32.W)), 1 -> Reg(UInt(32.W)))
// Gives
myMap_0
myMap_1
```
This is simple and makes sense, likewise:
```scala
val myMap = Map("foo" -> Reg(UInt(32.W)), "bar" -> Reg(UInt(32.W)))
// Gives
myMap_foo
myMap_bar
```
So far so good
```scala
val myMap = Map(someObj -> Reg(UInt(32.W)), anotherObj -> Reg(UInt(32.W)))
// Gives ???
myMap_ObjClass@23456789
myMap_ObjClass@deadbeef
// Or
myMap_0
myMap_1
```
And you can imagine how it gets more complicated from there. What about `Map[Data, Data]`??

shunshou, 2017-03-23:
Can you not limit to the first element of the tuple being a string or
number? My alternative was to create a wire with CustomIndexedBundle (my
version of your ProgrammaticBundle example) and assign... But having to do
that every time adds a lot of, imo, boiler plate.

ducky64, 2017-03-23:
We're already special-casing by handling Iterable containers, so limiting keys to strings or numbers isn't terrible (and addresses a common use case). Practically, anything that converted to a string is static between runs (i.e. no memory addresses) should be fine as a key. And even non-static strings might be fine, the generated FIRRTL just becomes nondeterministic...

It might also be worth refactoring the container naming, so the same code should be shared between annotation naming and the reflection naming.

ducky64, 2017-03-23:
(for the above refactor, it would be ideal to wait for the module hierarchy refactor to go in first, since that re-structures the reflection naming, and I want to avoid more merge nightmares from conflicting refactorings)

ducky64, 2017-03-23:
That being said, there's pretty much a consensus on the module hierarchy refactor, so assuming nothing else major comes up, it should go in after @sdtwigg reviews the updates. This fix should also be pretty fast, under a day's work.

shunshou, 2017-03-24:
Last comment (hopefully) @chiselName derps if you have 2 methods with the same name (different input types) in the Module.

ducky64, 2017-03-24:
Can you give me an example?

shunshou, 2017-03-24:
I'm actually not sure what it was complaining about anymore:

I had:

```
def context_complex_scalar_*(a: DspComplex[T], b: T, mode: Scalar):
DspComplex[T] = {
      val negB =
        DspContext.alter(context.copy(overflowType = dsptools.Grow,
numAddPipes = 0)) {
          Ring[T].zero context_- b
        }
      mode match {
        case ScalarReal => DspComplex.wire(a.real context_* b, a.imag
context_* b)
        case ScalarImag => DspComplex.wire(a.imag context_* negB, a.real
context_* b)
      }
    }
    def context_complex_scalar_reconfig_*(a: DspComplex[T], b: T, bIsImag:
Bool): DspComplex[T] = {
      val negB =
        DspContext.alter(context.copy(overflowType = dsptools.Grow,
numAddPipes = 0)) {
```

(The latter w/o _reconfig) and it wasn't compiling, but somehow mentioning :

```

 method context_complex_scalar_* is defined twice

[error]   conflicting symbols both originated in file
'/Users/angie.wang/Chisel3FFAST/FFTGen/src/main/scala/Chisel3/Chisel_WFTA.scala'

[error] @chiselName

[error]  ^

[error] one error found

```

Not 100% sure they're related. Moved on by renaming...

ducky64, 2017-03-24:
Yeah, if it was solved by renaming, it might not be a `@chiselName` annotation issue. One way to find out whether something is a `@chiselName` issue is to see if the error persists after removing the annotation - and if it does please file a bug...