chipsalliance/chisel #557: FixedPoint Mux (with different binary points)

shunshou, 2017-03-24:
That should be legal as long as the binary point values are "known", which is most of my use case. It's really annoying to write more complicated DSP b/c this isn't supported.

ducky64, 2017-03-24:
So the issue here is that as far as a Vec is concerned, `typeEquivalent` is strict - including checking that UInts are the same width. But the Mux constructor (which uses `cloneSupertype`) is a slightly relaxed case, using `typeEquivalent` except for `UInt` and `SInt` types which are special-cased. That special-casing does not handle FixedPoint, but that could be added (and should be safe as long as Bits and its children are sealed to chisel3 - but if that changes, we will need to introduce an extension API for Mux equivalency).

shunshou, 2017-03-24:
Is it possible to add it today-ish? I'm ok switching to some other mode of operation later, but this is making it really gosh darn awful to use with FixedPoint. To the end user, whatever you do internally doesn't matter... But if you did it for SInt, it *should* be done for FixedPoint.

ducky64, 2017-03-24:
Yeah, looking into it now.

ducky64, 2017-03-24:
Also, is this just FixedPoint, or is it cropping up in DspComplex (where it's a Bundle of two FixedPoint types)?

ducky64, 2017-03-24:
While the FixedPoint non-equivalent-binary-points can be solved by creating a new FixedPoint with the best precision and largest represented range, addressing the same issue when the FixedPoints are inside a Bundle is much more difficult.

shunshou, 2017-03-24:
I forget... I had a lot of Mux related issues last night/this morning. ~_~ 
But I did end up special casing a Mux just for DspComplex, where I separately did the muxing on the real/imag parts and then connected them back to a DspComplex output. Probably if I did that, I was running into similar issues...

ducky64, 2017-03-24:
If you special case a DspComplex mux (and are getting `can't create ... with FixedPoint with differing binaryPoints` rather than `can't create ... with non-equivalent types` error) then allowing differing BP FixedPoint should work.

shunshou, 2017-03-24:
Sure, it got to FixedPoint stuff, but I special cased Complex b/c I was having issues with the Mux type matching thing (the two Complexes were different types...)

ducky64, 2017-03-24:
Fixed fix is up

PR #559 (2017-03-24): Fixed fix, allow Mux of different binary points and widths
Resolves #557 

The implementation is ugly, but I can't think of a better way and this is a feature that makes sense. The larger problem seems to be that Chisel is duplicating width inferencing work with FIRRTL, but without all the information. The larger discussion might be whether that is a good practice, as removing width propagation in Chisel can remove a lot of complexity and special cases.
closed 2017-03-24 by ducky64