chipsalliance/chisel #565: More mux woes

shunshou, 2017-03-29:
I'm not really sure what's going on, but I found an alternative, so I'm posting this as an issue to be looked at later. 

When I mux a DspReal and zero (where DspReal is a bundle with a UInt node), I get:

```
[info]   firrtl.passes.PassExceptions: firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Chisel_WFTA.scala 391:17]: [module WFTA] Reference C72 is not declared.
[info] firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Chisel_WFTA.scala 391:17]: [module WFTA] Reference scalarZero is not declared.
[info] firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Chisel_WFTA.scala 392:17]: [module WFTA] Reference C73 is not declared.
[info] firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Chisel_WFTA.scala 392:17]: [module WFTA] Reference scalarZero is not declared.
[info] firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Chisel_WFTA.scala 405:17]: [module WFTA] Reference C76 is not declared.
[info] firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Chisel_WFTA.scala 405:17]: [module WFTA] Reference scalarZero is not declared.
```

The DspReal was type parameterized. When I used FixedPoint instead, it worked... not sure why Mux is so grumpy...

jackkoenig, 2017-04-18:
Can you provide the code causing it? Mux usually requires the arguments to be of the same type so I'm not sure what's going on here.

shunshou, 2017-04-18:
Remind me after tapeout...

ducky64, 2017-11-08:
I assume your tapeout is done now, so poke?

shunshou, 2017-11-13:
I'll have to look at it later, but I think it was Lit Bundle issue... I think a significant % of my problems with tapeout stemmed from the fact that I couldn't construct a Bundle of Lits and it wouldn't explicitly error...

ducky64, 2017-12-13:
This issue has been a zombie for a while, if it's still an issue, can you re-open this with an update?

closed 2017-12-13 by ducky64