chipsalliance/chisel #567: withClock and friends doesn't apply to child modules correctly

grebe, 2017-03-31:
In a module I essentially have the following

```
withClockAndReset(clock = newClock, reset = newReset) {
  val mod = Module(new ModuleOfSomeVariety)
  ...
}
```

and the module doesn't get newClock or newReset. According to @ducky64, setupInParent gets the parents clock whereas withClockAndReset only modifies the Builder's clock.

jackkoenig, 2017-03-31:
How embarrassing. I've PRed a fix

shunshou, 2017-03-31:
Hm I think this is sort of what I was trying to get at a couple of nights ago. It only worked w/ registers. Either way, I find this to be a little confusing...

jackkoenig, 2017-03-31:
@shunshou Yeah I definitely misunderstood you. What do you find confusing about the actual intended behavior?

```scala
val inst = Module(new MyModule)
inst.clock := someClock
inst.reset := someReset
// is replaced by
val inst = withClockAndRest(someClock, someReset) { Module(new MyModule) }
```

grebe, 2017-03-31:
Thanks so much @jackkoenig, you da real mvp

shunshou, 2017-03-31:
Like idk... to some extent, I think if you're overriding module clocks, doing the override_clock thing seems more explicit. Vs. if you wanted to override clks of individual registers, you'd consider using withClockAndReset. As you said, similar functionality, different implementation... but it'd be nice to make it clear exactly what the difference in implementation is.

commit c5ce93dd (2017-03-31): Make Module instantiations draw clock from Builder instead of parent

Fixes #567
PR #568 (2017-04-03): Make Module instantiations draw clock from Builder instead of parent
Fixes #567
commit bc05b7da (2017-04-03): Make Module instantiations draw clock from Builder instead of parent (#568)

Fixes #567
closed 2017-04-03 by aswaterman