chipsalliance/chisel #569: Vec.fill does not work as expected

MahdiNazemi, 2017-04-02:
I had a working code written a few weeks ago which uses Vec.fill as follows:

`val multipliers = Vec.fill(n)(Module(new FPMul()).io)`

The same code doesn't work today neither using FIRRTL nor Verilator. 

Do I need to change something to make the code work again?

Thanks,
Matt

jackkoenig, 2017-04-02:
Hi Matt,

Can you provide a little more information about what is not working? Are you doing dynamic indexing on the multipliers and it's not working? I am looking into the issue and I believe I am seeing a problem with dynamic indexing

MahdiNazemi, 2017-04-02:
This is the code I used to implement a fully parallelized vector-vector outer product:

```
  val multipliers = Vec.fill(n * n)(Module(new FPMul()).io)

  for (i <- 0 until n) {
    for (j <- 0 until n) {
      multipliers(i * n + j).a := io.a(i)
      multipliers(i * n + j).b := io.b(j)
      io.out(i * n + j) := multipliers(i *n + j).out
    }
  }
```

The values returned during unit testing of io.out, which is an n-by-n matrix, are all zeros.

As I mentioned earlier, the Verilog code is okay and the indices are as expected.

sdtwigg, 2017-04-03:
Unless you need the dynamic indexing of Vec somewhere else, what happens if you use Seq instead?

This will avoid the creation of an intermediary wire (although this shouldn't have mattered in this case...).

jackkoenig, 2017-04-03:
This appears to be broken by 375e2b6a0a456c55298d82837d28986de6211ebc but not adequately fixed by a6798adbf7567cf105f8190ebd293a4cfb3aeb11

jackkoenig, 2017-04-03:
A simple test case:

```scala
class TestCase extends BasicTester {
  val w = Wire(UInt(32.W))
  val mods = Vec.fill(2)(Module(new PassthroughModule).io)

  for (i <- 0 until 2) {
    mods(i).in := 123.U
    w :=  mods(i).out
  }
}
```

Before 375e2b6a0a456c55298d82837d28986de6211ebc:
```
    wire mods : {flip in : UInt<32>, out : UInt<32>}[2] 
    mods is invalid 
    mods[0].out <= PassthroughModule.io.out 
    PassthroughModule.io.in <= mods[0].in 
    mods[1].out <= PassthroughModule_1.io.out 
    PassthroughModule_1.io.in <= mods[1].in 
    mods[0].in <= UInt<7>("h07b") 
    w <= mods[0].out 
    mods[1].in <= UI
```
After
```
    wire mods : {flip in : UInt<32>, out : UInt<32>}[2] 
    mods is invalid 
    mods[0].out <= PassthroughModule.io.out 
    mods[0].in <= PassthroughModule.io.in 
    mods[1].out <= PassthroughModule_1.io.out 
    mods[1].in <= PassthroughModule_1.io.in 
    mods[0].in <= UInt<7>("h07b") 
    w <= mods[0].out 
    mods[1].in <= UInt<7>("h07b") 
    w <= mods[1].out
```

jackkoenig, 2017-04-03:
Same problem as what motivated #522 and my branch illustrating that problem (https://github.com/ucb-bar/chisel3/tree/vec-of-modules) is still failing

jackkoenig, 2017-04-03:
Looking at the emitted Firrtl above, the wire actually has the right directions, it's just that the connections are wrong.

MahdiNazemi, 2017-04-04:
> Unless you need the dynamic indexing of Vec somewhere else, what happens if you use Seq instead?

It works well.

commit a6a54410 (2017-04-03): Use input element to decide if Vec of values has direction

Using the sample_element of the created wire is incorrect because Wires have no
direction so the Wire constructed for a Vec of Module IO was constructed
incorrectly. Fixes #569 and resolves #522.
PR #570 (2017-04-04): Use input element to decide if Vec of values has direction
Using the sample_element of the created wire is incorrect because Wires have no
direction so the Wire constructed for a Vec of Module IO was constructed
incorrectly. Fixes #569 and resolves #522.

Once I figured out that this was the problem, I was confused how you could index into a wire that was of type Vec of Module io and still correctly bulk connect to the directioned inputs. It turns out that this works by using the direction of the other side of the bulk connect as a hint. This has the following two nasty consequences that shoul [... cut]
commit 00796dfc (2017-04-04): Use input element to decide if Vec of values has direction (#570)

Using the sample_element of the created wire is incorrect because Wires have no
direction so the Wire constructed for a Vec of Module IO was constructed
incorrectly. Fixes #569 and resolves #522.
closed 2017-04-04 by jackkoenig