chipsalliance/chisel #576: RegEnable doesn't generate a register when enable bit is tied to false.B

jchang0, 2017-04-11:
I would like this to generate the equivalent verilog code:

  always @(posedge clock) begin
    last_value_reg <= last_value_reg;
  end

The usage case is that I want an assignment to be assigned one random value in the beginning, then retain its value throughout the simulation.

However, this chisel code optimizes the register away, and the last_value_reg becomes a floating signal wire, which actually don't work for my flow.

class TestCircuit extends Module {
    val io = IO(new Bundle {})

    val float_value = Wire(UInt(1.W))
    val last_value_reg = chisel3.util.RegEnable(float_value, false.B)
}

It seems like this is a workable workaround for the time being though:

class TestCircuit extends Module {
    val io = IO(new Bundle {})

    val float_value = Wire(UInt(1.W))
    val enable = Wire(Bool())
    enable := false.B
    val last_value_reg = chisel3.util.RegEnable(float_value, enable)
}

mwachs5, 2017-06-01:
Should we just create a primitive that is "random assigned value"?

mwachs5, 2017-06-01:
I mean a library component vs something in chisel

jackkoenig, 2017-06-01:
We have [LFSR16](https://chisel.eecs.berkeley.edu/api/index.html#chisel3.util.LFSR16$) in chisel utils. Larger lfsrs might be helpful here?

Also per the original issue, not sure about what's happening with the RegEnable, that looks like a bug, but as you pointed out in #621, just a standard Reg with no reset or connections should fulfill your use case.

jchang0, 2017-06-01:
What I wanted is a random value that doesn't change from cycle to cycle.  #621 also have the same issue where the register is optimized away.

ducky64, 2017-11-08:
Random sources are outside the domain of RTL construction, and should be blackboxed. This can be composed with a Chisel Reg if needed for stability.

closed 2017-11-08 by ducky64