chipsalliance/chisel #577: Gender issue

chiselmonster, 2017-04-12:
How come this works
```
io.outfinal :=  a3.io.out ## a2.io.out ## a1.io.out ## a0.io.out
```
but not this
```
  io.outfinal(3) :=  a3.io.out
  io.outfinal(2) :=  a2.io.out
  io.outfinal(1) :=  a1.io.out
  io.outfinal(0) :=  a0.io.out
```
the full code is as such

```
class Adder extends Module {
  val io = IO(new Bundle {
    val in0 = Input(UInt(1.W))
    val in1 = Input(UInt(1.W))
    val carryin = Input(UInt(1.W))
    val carryout = Output(UInt(1.W))
    val out = Output(UInt(1.W))
    val test = Output(UInt(2.W))
  })

  val add = RegInit(UInt(2.W), 0.U)
  add := io.in0 +& io.in1 +& io.carryin
  io.out := add(0)
  io.carryout := add(1)
  io.test := add
}

class FullAdder extends Module {
  val io = IO(new Bundle {
    val in0 = Input(UInt(4.W))
    val in1 = Input(UInt(4.W))
    val carryin = Input(UInt(1.W))
    val carryout = Output(UInt(1.W))
    val outfinal = Output(UInt(4.W))
    val test = Output(UInt(5.W))
  })

  val a0 = Module(new Adder())
  a0.io.carryin := io.carryin
  a0.io.in0 := io.in0(0)
  a0.io.in1 := io.in1(0)

  val a1 = Module(new Adder())
  a1.io.carryin := a0.io.carryout
  a1.io.in0 := io.in0(1)
  a1.io.in1 := io.in1(1)

  val a2 = Module(new Adder())
  a2.io.carryin := a1.io.carryout
  a2.io.in0 := io.in0(2)
  a2.io.in1 := io.in1(2)

  val a3 = Module(new Adder())
  a3.io.carryin := a2.io.carryout
  a3.io.in0 := io.in0(3)
  a3.io.in1 := io.in1(3)

  //io.outfinal :=  a3.io.out ## a2.io.out ## a1.io.out ## a0.io.out
  io.carryout :=  a3.io.carryout
  io.outfinal(2) :=  a2.io.out
  io.outfinal(1) :=  a1.io.out
  io.outfinal(0) :=  a0.io.out

}
```
The error msg is as follows  :
```
[error] firrtl.passes.PassException: 3 errors detected!
firrtl.passes.PassExceptions:
firrtl.passes.CheckGenders$WrongGender:  @[Adder.scala 64:18:@116.4]: [module Fu
llAdder]  Expression _T_22 is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[Adder.scala 65:18:@118.4]: [module Fu
llAdder]  Expression _T_23 is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[Adder.scala 66:18:@120.4]: [module Fu
llAdder]  Expression _T_24 is used as a FEMALE but can only be used as a MALE.
firrtl.passes.PassException: 3 errors detected!
        at firrtl.passes.Errors.trigger(Passes.scala:31)
        at firrtl.passes.CheckGenders$.run(Checks.scala:563)
        at firrtl.SimpleRun$$anonfun$runPasses$1$$anonfun$3.apply(Compiler.scala
:144)
        at firrtl.SimpleRun$$anonfun$runPasses$1$$anonfun$3.apply(Compiler.scala
:144)
        at firrtl.Utils$.time(Utils.scala:135)
        at firrtl.SimpleRun$$anonfun$runPasses$1.apply(Compiler.scala:144)
        at firrtl.SimpleRun$$anonfun$runPasses$1.apply(Compiler.scala:141)
        at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized
.scala:124)
        at scala.collection.immutable.List.foldLeft(List.scala:84)
        at firrtl.SimpleRun$class.runPasses(Compiler.scala:141)
        at firrtl.PassBasedTransform.runPasses(Compiler.scala:163)
        at firrtl.PassBased$class.runPasses(Compiler.scala:159)
        at firrtl.PassBasedTransform.runPasses(Compiler.scala:163)
        at firrtl.PassBasedTransform.execute(Compiler.scala:167)
        at firrtl.Compiler$$anonfun$8$$anonfun$9.apply(Compiler.scala:323)
        at firrtl.Compiler$$anonfun$8$$anonfun$9.apply(Compiler.scala:323)
        at firrtl.Utils$.time(Utils.scala:135)
        at firrtl.Compiler$$anonfun$8.apply(Compiler.scala:323)
        at firrtl.Compiler$$anonfun$8.apply(Compiler.scala:321)
        at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized
.scala:124)
        at scala.collection.immutable.List.foldLeft(List.scala:84)
        at firrtl.Compiler$class.compile(Compiler.scala:321)
        at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:137)
        at firrtl.Driver$.execute(Driver.scala:167)
        at chisel3.Driver$.execute(Driver.scala:161)
        at chisel3.iotesters.setupFirrtlTerpBackend$.apply(FirrtlTerpBackend.sca
la:110)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply$mcZ$sp(Driver.scala
:48)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:35)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:35)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at chisel3.iotesters.Driver$.execute(Driver.scala:35)
        at chisel3.iotesters.Driver$.apply(Driver.scala:212)
        at hello.FullAdder$.main(Adder.scala:94)
        at hello.FullAdder.main(Adder.scala)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
[trace] Stack trace suppressed: run last compile:run for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
        at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last compile:run for the full output.
[error] (compile:run) Nonzero exit code: 1
[error] Total time: 6 s, completed 12 Apr, 2017 10:24:42 AM
```

ducky64, 2017-04-12:
Subword assignment (assigning to a bit slice of a Bits type, like UInt) isn't (yet?) supported.
Unfortunately, the error message for this limitation isn't really helpful. See #399.

You can, however, assign to sub-elements of a Vec or Bundle. You can also Cat bits together..

chiselmonster, 2017-04-12:
Indeed Cat or ## managed to resolve this issue, but since this command has not altered the direction of the pins, I am wondering if there is a cleaner solution instead of a current workaround

ducky64, 2017-04-12:
So our view is that subword assignment is an unnecessarily low-level construct for uses better served by having structured Bundles or Vecs - instead of assigning to bitslices with indices you need to manually compute, it makes more sense to assign to elements in a Bundle. Note that Bundle does define bit packing order, and it's statically known if all elements are fixed width.

Unfortunately, the error message here isn't helpful. nor does it document our intent. It's on our list of improvements...

As for your example, we usually prefer writing Chisel generators rather than fixed-parameter modules (like a n-wide full adder vs. a 4-bit full adder), and would probably use Cat on a list rather than individual bit assignments.

closed 2017-04-12 by ducky64