chipsalliance/chisel #579: FIRRTL Pass from Chisel

ssbanerje, 2017-04-14:
How do I add a custom pass for the FIRRTL compiler from a Chisel3 project? I have been looking through the [API](https://chisel.eecs.berkeley.edu/api/) and can't seem to find any way to do this.

Do I have to fork a separate Chisel3 and FIRRTL repo add the pass there and compile my circuit using the forked Chisel repo?

azidar, 2017-04-14:
You don't need to fork FIRRTL or Chisel3 to add a custom pass. If you can take a look at the tutorial in FIRRTL as an example of how to insert a custom pass to the FIRRTL compiler, see the FIRRTL README for instructions: https://github.com/ucb-bar/firrtl.

ssbanerje, 2017-04-15:
@azidar Maybe I am missing something, but that seems to give instructions only for transforms, i.e., using the `--custom-transforms` flag. Are `firrtl.Transform` and `firrtl.passes.Pass` equivalent?

I am following the [FIRRTL wiki page](https://github.com/ucb-bar/firrtl/wiki/Common-Pass-Idioms) that describes `Circuit` modifications.

azidar, 2017-04-19:
Yes, Pass is a subtype of Transform, this is a recent change and the wiki hasn't been updated yet.

closed 2017-11-08 by jackkoenig